Ultimaker/Cura #6579: Running Cura with --debug crashes because sys.stderr is None

fieldOfView, 2019-10-25:
**Application version**
4.3.0 (releases, frozen)

**Platform**
Windows 10

**Printer**
n/a

**Reproduction steps**
Open a prompt
cd to installed Cura location
run Cura with --debug argument `cura.exe --debug`

**Actual results**
Cura crashes with the following traceback:
```
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "Console.py", line 24, in <module>
  File "X:\4.3\build\inst\bin\cura_app.py", line 134, in <module>
RuntimeError: sys.stderr is None
```

**Expected results**
Cura runs and log output is shown on the console

**Additional information**
n/a

smartavionics, 2019-10-25:
You can use cura.exe --debug OK from the bash shell that comes with git.

Ghostkeeper, 2019-10-25:
I've made a workaround for you. It still works for me. Can you see if that fixed it?

closed 2019-10-25 by Ghostkeeper

fieldOfView, 2019-10-25:
@smartavionics doesn't work for me with the "git bash" that I found on my system

@Ghostkeeper, I'll give it a go over the weekend (I don't have a build system for Windows set up, so I'll have to rely on some manual patching)

commit 0168a1d5 (2019-10-25): Don't output to stderr if there is no stderr

This can happen on Windows where the default command line doesn't have a stderr channel. Put it in stdout then.

Fixes #6579.