Ultimaker/Cura #6584: Launch Cura 4.3 in Windows 10 - Stays as a background app and never changes to running app.

nicholsaz, 2019-10-25:
**4.3**
(The version of the application this issue occurs with.)

**Windows 10**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)

**Ender 3**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)

**Reproduction steps**
1. Something you did.
2. Something you did next.

**Launch App from Taskbar**
(becomes background app and never a running app (when viewed from task manager).)

**Cura opens**
(What should happen after the above steps have been followed.)

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

rburema, 2019-10-29:
Hi @nicholsaz 
 - Does this occur with older versions of Cura as well, or just 4.3?
 - Do you use a laptop, and if so, do you use external screens with some sort of adapter? If not, are you trying to start Cura in a remote session?
 - What graphics card is Cura using?
 - Are all drivers (especially graphics cards ones) up to date?
cheers,
Remco.

nicholsaz, 2019-10-30:
It seems to only be occurring with 4.3. I have all version still installed.
It’s being started as I normally have all versions - pegged to the taskbar (and not) on my laptop.
Not running a remote session.
The laptop uses an Intel HD 520 graphics chipset.

If I run the app as an administrator it (usually) starts fully. Otherwise, it stays as a background app and never comes into focus.

Thanks,

Mark

rburema, 2019-10-30:
Since you start it as an administrator sometimes, maybe the .lock file stays on disk for some reason? It's in `%appdata%/cura/<version>`, and it should have been deleted after startup.

Otherwise, it seems like an issue we're working on. The problem is that it seems to have a lot of different causes. Some users have fixed it for themselves, see: https://community.ultimaker.com/topic/29861-cura-wont-open-in-windows-10-i-may-have-a-solution

But this doesn't solve everyones issue (see also here: https://github.com/Ultimaker/Cura/issues/6197 )

We do have some potential fixes, but no guarantee that those'll work 'in the wild'. Keep an eye out for the 4.4-BETA later on if the problem persists.

dstnceswmer, 2021-02-13:
Wondering if there is any more updates on this? I just experienced this for the first time yesterday with version 4.8.
Things I've tried:
Uninstalling/reinstalling after deleting all appdata cura folders (local and roaming)
Downgrading to 4.7 and 4.4
Unplugging 2nd and 3rd monitors
Updating GPU drivers
Uninstalling/reinstalling the VisualC++ redistributables 2015 and later
Manually ran vcredist_x64.exe from the cura install directory
Allowed in Windows defender firewall
Have VPNs turned off
Tried launching multiple instances and let sit
Compatibility mode to Windows 8
Run as admin manually (right click menu)

Every time it will appear as a background process in task manager but never boots up as an application running.
I can provide any logs requested (As long as I'm told where they are/how to get them :) )

[image]

rburema, 2021-02-16:
@dstnceswmer You can find  the logs if you click Help -> Show Configuration Folder. It should show up there, as `cura.log`. If you have any difficulties in attaching the file to this issue, please either zip it or rename it.

It might also help to take a look at `cura.cfg`, at least if you switch working configurations often (like the number or monitors attached to the system running Cura).

GregValiant, 2024-12-15:
Is this still an issue with current Cura versions(5.8 and up)?  Can this be closed?

closed 2024-12-30 by github-actions[bot]