Ultimaker/Cura #6585: Unsupported small features

BagelOrb, 2019-10-25:
This is a general problem of how we handle support and infill.
Cura designates areas to fill with support and filling it with some density of support structure, but that approach fails for small overhang protrusions.

There is no easy fix for this. This issue can only be solved by rethinking our infill and support systems. I don't expect this issue to be closed any time soon.

Example for infill:
The skin generated for these dice dot cutouts is wholly unsupported and creates stringing:
[image]
[dice_models.zip](https://github.com/Ultimaker/Cura/files/3773116/dice_models.zip)

Example for support:
Because of the Z distance an outline will get generated, but still those outlines are unsupported.
[image]
[axes_models.zip](https://github.com/Ultimaker/Cura/files/3773128/axes_models.zip)

related: https://github.com/Ultimaker/Cura/issues/5440
credits: https://www.reddit.com/r/3Dprinting/comments/dmzib2/prevent_filter_or_warn_about_extrusions_into_thin/

BagelOrb, 2019-10-25:
**Physical problem:**
Such unsupported regions causes spaghetti which will get wiped off on some unknown location. It's undefined printer behavior. Who knows what mischief the spaghetti might cause.

Moreover, the missing print feature will cause defects on further layers above.

BagelOrb, 2019-10-25:
**Impact**
I believe that small holes and small protrusions in horizontal faces are quite common in engineering applications.

Ghostkeeper, 2019-10-30:
Duplicate of https://github.com/Ultimaker/Cura/issues/6001

closed 2019-10-30 by Ghostkeeper