Ultimaker/Cura #6586: no nozzle support 0.1mm or more than 1mm in the printer profile Creality Ender3

Ultraruss, 2019-10-25:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)

**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)

**Reproduction steps**
1. Something you did.
2. Something you did next.

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

Ultraruss, 2019-10-25:
no nozzle support 0.1mm or more than 1mm in the printer profile Crality Ender3 or CR10

rburema, 2019-10-29:
Hi @Ultraruss When it comes to 3rd party printers, we're wholly dependant on community contributions. If someone makes a profile that has <= 0.1mm or > 1mm for that type of printer, we'll be happy to merge it if they post a PR here, but until that time, we can't do much.

Ghostkeeper, 2019-10-30:
We'll close this since there's nothing we can really help you with. Pull requests to add profiles for 0.1mm or >1.0mm nozzles are welcome! So if you feel like tuning some profiles, go ahead and let us know what settings you think should be used.

closed 2019-10-30 by Ghostkeeper