Ultimaker/Cura #6587: Cura is not printing in the middle of the print bed

bradley1234567, 2019-10-25:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)

**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)

**Reproduction steps**
1. Something you did.
2. Something you did next.

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

rburema, 2019-10-29:
Does this show up in the preview (does it slice), or is it only a problem once the printer starts?

fieldOfView, 2019-10-29:
In Machine Settings, is the setting "Origin at center" checked or not? It should *NOT* be checked.

rburema, 2019-10-30:
Hi @bradley1234567 
Did you uncheck 'origin at center' like @fieldOfView asked?

In any case: If the fault is not with the printer itself (it might be), then we can perhaps learn something from a project file. Could you save (_not_ export) one of the projects this goes wrong with and attach it here? (You might want to zip or rename the extension to get github to accept the attachment.)

cheers, Remco.

bradley1234567, 2019-10-30:
Yes origin was unchecked

closed 2020-06-19 by no-response[bot]