Ultimaker/Cura #6589: OSX UI performance?

tablatronix, 2019-10-27:
**Application version**
4.3.0

**Platform**
osx macbook

**Printer**
ender 3

There seems to some UI performance issue on osx which is presented as lag, settings pane takes a while to load up, and right clicking always takes 1-2 seconds to open.

It has been a problem for as long as I have used cura and can make some things excruciating.

I have disabled plugins, does anyone else experience this ?

Ghostkeeper, 2019-10-28:
So far at work we haven't seen any particular outliers with MacOS except the duration of the slicing process itself, which had to stay single-threaded up until recently due to our compiler of choice.

We've made a few major changes that affect performance in version 4.4. In particular, we've added a feature that made some things slower, and then we made a performance tweak that made many things faster. I believe the current state makes uninstalling plug-ins and deleting profiles much slower, switching profiles and printers much faster, and the rest within standard deviation the same. But we can expect things to be a little different in the version the developers are working with now. I think we should wait with this until the beta for 4.4 is released.

fieldOfView, 2019-10-28:
Having said that, it never hurts to have a look at your Cura.log to see if there's anything "funny" going on that makes your experience extra laggy. That file can be found through the Help -> Show configuration folder. Please upload your cura.log file somewhere and post a link here.

tablatronix, 2019-10-28:
I did a test it seems to be related to the number of options in the sidebar, on show all settings, the menu takes a while to draw on start and UI interaction is slow, If I collapse or switch to basic, it gets speedier. So something in the panel display is probably causing too many updates or doing something in qt. 

Here are some errors , nothing glaring..

```PHP
2019-10-28 06:12:10,863 - ERROR - [Thread-3] UM.Logger.logException [84]: Exception: Job <USBPrinting.AutoDetectBaudJob.AutoDetectBaudJob object at 0x600017df9d98> caused an exception
2019-10-28 06:12:10,878 - ERROR - [Thread-3] UM.Logger.logException [88]: Traceback (most recent call last):
2019-10-28 06:12:10,891 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/Users/ultimaker/build/env/4.3/inst/lib/python3.5/site-packages/serial/serialposix.py", line 501, in read
2019-10-28 06:12:10,909 - ERROR - [Thread-3] UM.Logger.logException [88]: serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2019-10-28 06:12:10,917 - ERROR - [Thread-3] UM.Logger.logException [88]: 
2019-10-28 06:12:10,922 - ERROR - [Thread-3] UM.Logger.logException [88]: During handling of the above exception, another exception occurred:
2019-10-28 06:12:10,938 - ERROR - [Thread-3] UM.Logger.logException [88]: 
2019-10-28 06:12:10,951 - ERROR - [Thread-3] UM.Logger.logException [88]: Traceback (most recent call last):
2019-10-28 06:12:10,960 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/Users/ultimaker/build/4.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 121, in run
2019-10-28 06:12:10,973 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/AutoDetectBaudJob.py", line 74, in run
2019-10-28 06:12:10,988 - ERROR - [Thread-3] UM.Logger.logException [88]:     line = serial.readline()
2019-10-28 06:12:10,999 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/Users/ultimaker/build/env/4.3/inst/lib/python3.5/site-packages/serial/serialposix.py", line 509, in read
2019-10-28 06:12:11,012 - ERROR - [Thread-3] UM.Logger.logException [88]: serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2019-10-28 06:12:13,136 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 12.3s
2019-10-28 06:12:17,953 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2019-10-28 06:12:18,733 - INFO - [MainThread] cura.Machines.MaterialManager._updateMaps [281]: Updating material lookup data ...
2019-10-28 06:12:18,818 - ERROR - [MainThread] cura.Machines.MaterialManager.__addMaterialMetadataIntoLookupTree [266]: Material custom_material_imade3d_jellybox_0.4_mm_2-fans contains a variant 0.4 mm 2-fans that does not exist. It will not be added into the material lookup tree.
2019-10-28 06:12:18,848 - ERROR - [MainThread] cura.Machines.MaterialManager.__addMaterialMetadataIntoLookupTree [266]: Material generic_pla_175 #2_imade3d_jellybox_0.4_mm_2-fans contains a variant 0.4 mm 2-fans that does not exist. It will not be added into the material lookup tree.
```

fieldOfView, 2019-10-28:
Make a set of visible settings you really use daily, instead of making everything visible. Your life will be better for it (once you get used to it). 

Unless you use USB printing, you might want to disable the USB Printing plugin through the "Installed" tab in the Marketplace,

tablatronix, 2019-10-28:
good tips, I seem to be having much better performance on my newer macbook, so maybe my computer is just crap.. Might try wiping it a trying a fresh install also.

Will close later depending on what I find.

nallath, 2019-10-28:
If all goes well, the fixes I made in https://github.com/Ultimaker/Cura/pull/6583 will be merged into 4.4. Almost all of these changes are aimed at making the UI faster.

tablatronix, 2019-10-29:
k will definitely beta or alpha test

fieldOfView, 2020-03-01:
Has this improved in 4.4 and 4.5?

tablatronix, 2020-03-01:
Nope

fieldOfView, 2020-03-01:
Thanks for beta testing.

tablatronix, 2020-03-01:
lol, Sorry not very helpful, is there some new enabled logging I can post? Some things seems a little faster, but the majority of issues like all UI interaction are lagged still, simply changing the color legends takes several seconds , you click and have to wait for it to pop up, same with most stuff dropdowns etc..

Ghostkeeper, 2020-03-06:
In 4.5 the most major performance improvement to the UI is that it doesn't re-render the scene upon re-rendering the UI. So if you're in layer view or you have lots of big models loaded I'd expect the UI to be more responsive.

tablatronix, 2021-05-29:
It has gotten even worse with the past 2 updates, I can't barely use it now. Shrug. Let me know if there are dev or test stuff for feedback required.

tablatronix, 2021-06-06:
Ok I have confirmed a few times that this is the case, If I open cura and start using it in windowed mode, it barely functions. I have to resize it and then it works. So something is initializing wrong with the rendering and is reseting when you resize..

Ghostkeeper, 2021-06-07:
That's weird, yeah. We're not really doing anything special upon first initialisation vs. other renders. When you change the window size, Qt automatically re-renders the UI. And Cura will trigger the 3D scene to be redrawn. The 3D scene redrawing happens with any change to the scene as well (rotating the camera, loading a new model, etc). The UI redrawing doesn't happen with any change to the scene, but that's also completely out of our hand and implemented in Qt, not in Cura's source code.

tablatronix, 2021-06-08:
Yeah rotating the camera is one of the things that works always.. I made a video ill throw it up or redo it with a overlay for the interaction

Ghostkeeper, 2021-06-15:
Is it just the clicking, or the frame rate? For example, if you load a big-ish STL file of 10-20MB, do you get a smooth animation with the bouncing progress bar in the bottom?

[image]

If the animation is smooth, then it's just the clicks that aren't registering fast enough. If the progress bar is barely bouncing, the entire rendering performance is bad.

However either way I wouldn't really know what to do about it. We just create `Button` objects and such and all the click handling is done by Qt. How would you tell Qt to get clicks faster? To improve performance we could look at reducing the number of UI elements and such but that's already pretty well optimised.

tablatronix, 2021-06-15:
Ill check that also