Ultimaker/Cura #6593: Ender-3 multiple extruder option

TheSin-, 2019-10-28:
I have an Ender-3 which I'm upgraded to use an MKS GEN-L and run 2 extruders.  In Cura there is no way to add a 2nd extruder, the option simply isn't there.  I modified all my local files and added a 2nd extruder but it was a ton of work.

Not I have a Mosaic Palette+ which turns one extruder into 4.  So I need to have 5 extruders, I can modify the files again but why can't all printers just have an option to add extruders, why is this locked?  Could it not just have a default value of 1 and allow up to 8 or something??

I'd very difficult to add extruders otherwise, and I'm sure not everyone has the OS experience to mod this files by hand.

fieldOfView, 2019-10-29:
>  Could it not just have a default value of 1 and allow up to 8 or something??

Sure. Start with a Custom FFF Printer, and you can choose up to ... 8 extruders!

nallath, 2019-10-29:
This is by design. Machine files define machines in a given (aka; factory) state. Some changes can be made to those, but adding extruders is a bit tricky to get right (as you have already noticed).
At some point, the number of changes made to a printer mean that it's a different (and thus custom) printer.

TheSin-, 2019-10-29:
I respectfully disagree.

1. a palette can be added to any stock printer and will require 4 extruders to make it work.

2. my printer is modified but it's still prints 100% like an ender 3, it's the same hotness, same extruders and same firmware.  To recreate what was added with the merge of creawesome mod would take far far longer.  And I have tried to make a custom it prints like crap comparatively.

Allowing multiple extruders isn't deviating from the factory print it's just allowing for addons like the palette or MMU2 or in my case a second extruder or chimera or cyclops.  Those are very simple mods and with the exception of the last 2 which use a v6 nozzle the hotends are the same so it prints the same.

Why limit the extruders? it's very little work to just allow multiples were as trying to copy the settings into a custom FFF printer is a HUGE task.  And it still won't print the same after weeks of tweaking.

I thought Cura was around to advance the 3d printing community not stifle it by shooting down every request for improvement and helps the community.

The Mosaic Palette a fun and great addition to and printer. and I just had a second extruder for PVA supports cause lets be honest support no matter how good ruins prints.  It's never as smooth under supports.   And just dropping a 1 down and select a 2 or 5 doesn't change how that first nozzle prints at all.

nallath, 2019-10-30:
It's not very little work for us to change this. If it were, I wouldn't be explaining why we didn't do it, I would have made the change already. 

I agree that the palette is a super cool addition to the printer, but that still doesn't decrease the amount of effort required to implement it. So yeah, Cura is around to advance 3D printing, but that doesn't mean that we can (or will) honor every feature request. I'm not sure if you intend it as such, but your message comes across as pretty passive agressive, which won't help convincing anyone that implementing what you ask is a good idea.

TheSin-, 2019-10-30:
You are right and I apologies, just seems every request I have gets the same response.  Anyhow I’ll continue to fight with it per release locally, I’m sure I can script it for myself at least, thanks anyhow.

closed 2019-10-30 by TheSin-

nallath, 2019-10-30:
We're simply understaffed for the number of feature requests that we get. So as much as I'd love to build all the things, we have to use pretty aggressive triage to decide what does (and does not) get done. So just because it doesn't get picked up by us, it doesn't mean that it's not a valid idea or feature (just that there are other things that get more priority)

TheSin-, 2019-10-30:
nod, just through tit'd be a nice easy win for you guys since you just have to copy the extruders/creatliy_base_extrude_0.def.json -> 1,2,3,4,5,6,7 then add those 7 new ones to the definitions/creality_base.def.json

For me I have to do that every time, then I have to delete my printer and recreate it, cause the app sees the changes and resets.  So every release I need to reset from scratch which is a PITA.

For you it's 7 new files and modding one and that's all then all creality printers can have 8 extruders with 1 being the default.  8 might be much I personally only need 5, but it doesn't hurt so I make 8 each time.

Anyhow keep up the good work and sorry I got snippy, I'm just frustrating which loosing all my settings every update and having to remember them all each time.

TheSin-, 2019-10-30:
for sure, but it's really a scriptable thing IMHO.  But it's okay just upsets me that the only way for me on update is to restart my settings, but I'm going to try on the next update to patch before I start it and hope I don't have to reset.  I think it's resets cause it noticed change.

I just thought it'd be a really good feature cause I know I get ppl asking me how to do it all the time and I really do not like telling ppl how to modify in the app it's self.

As for maintenance I might just be ignorant cause I can not thing of an instance of needed to, it's just extra extruders with no config, but then again I have yet to have to break or affect anything, once they are available Cura deals with them amazingly well, it's just a problem to get them in.

Regardless I already starts on a script to deal with it for be 4.4, hopefully the filenames don't change too much, but either way I'll sort it cause I like the creawesome results and don't want to loose that, and need 5 extruders, so I'll figure it out.

nallath, 2019-10-30:
We have to keep filenames pretty stable, because it wreaks havoc on our machines if it doesn't.