Ultimaker/Cura #6597: 4.3. awesomeness, but on-screen hints needed

FlailAway, 2019-10-29:
**Is your feature request related to a problem? Please describe.**
I think so, or I can't find the settings to fix it.

**Describe the solution you'd like**
Mint 19.2+Cinnamon
In the Prepare and Preview pages I'd like to see the loaded filename(s) constantly on screen.
Same for any/all Extensions being used in the **current** Slice.

I just sliced a thing which had a "Pause at Layer" and set it printing, I managed the Pause call at that layer and left it to complete normally over several hours. All good.

Just after I had managed the Pause of the first part, I sliced the next part, stupidly forgetting I had the Pause Extension still in play. I came back 4 hours later and it was waiting for a "press key," from me.

Am I missing some Settings somewhere?

rburema, 2019-10-30:
Regarding the first issue, 4.2 added the 'object list' in the lower left of the screen, above the view-point selection icons. Uncollapse the object list and you have the (base) filenames. Is this what you wanted?

When it comes to the second request, what would you suggest? I mean: where would we show all of that? IMHO: There are a _lot_ of similar things you can forget, if we put all of them on screen Cura might start looking like the cockpit of a large plane. Or do you think that in case of the the 'pause at layer' _specifically_ this is so (user) error-prone that it warrants its own warning?

FlailAway, 2019-10-30:
Issue #1: Ahhh, OK, I didn't click the little "up" chevron.
Issue #2: OK, point taken, but then what about a check box on the List that is auto-cleared after every print so it has to be selected? I'd much rather forgte to set that and have the print careen through where I thought it was going to to pause rather than Pause and wait forever. 

If I need a pause, then I am going to be there waiting for it. But if I have a 7-hour print, I do not want to come back the next day and find it still 10-minutes into the print and waiting for a "click," with the heaters on all that time. I'd better get a direct line put in to the local Fire Station. :)

Maybe it is just me, but to have an invisible option that applies to all slicing is a bit too brute force-ish and user unfriendly. What about having the "Extensions" menu item in bold+red if anything is in play?

It is _only_ error prone to the normal humans among us. ;)

rburema, 2019-10-31:
> but then what about a check box on the List that is auto-cleared after every print so it has to be selected?

> What about having the "Extensions" menu item in bold+red if anything is in play?

Those are nice suggestions. I can ask around and see if we have time to implement something like either of that.

Ghostkeeper, 2019-10-31:
I think an "alarm" icon for when a post-processing script is active is a bit overboard, as hopefully it will have been on purpose. Maybe we can think of an icon that is a bit better to recognise as a post-processing script than the current hammer and wrench?

FlailAway, 2019-10-31:
> ... though the first one also has the opposite problem; each time you adjust a setting after slicing, you'd have to re-check the required post-processing script.

True, but after using the Pause for a print yesterday when I went to load a sister part with no Pause, it occurred that check box(es) only be cleared when clearing the build plate.

FlailAway, 2019-10-31:
> I think an "alarm" icon for when a post-processing script is active is a bit overboard, as hopefully it will have been on purpose.

That is true for the print that requires the Pause. But my problem was forgetting that the Pause was still in play after I had cleared the build plate and loaded a different part. Then with no indication that the Pause was still in play I set it printing and went home.

Ghostkeeper, 2019-11-01:
> Then with no indication that the Pause was still in play

That's what Rburema meant: There is an indication. It's a hammer and wrench icon next to the slicing button.

FlailAway, 2019-11-01:
But, it does not clear when I Clear Build Plate. Surely if I am clearing and starting with a new file it would be HIGHLY unlikely that I'd want to have a Pause at the same layer/height. "Clear," should mean Clear everything I am loading a new object file. It makes sense to keep the current Slicing Profile but not specific things to a print. 

I give up and will stop trying to help straighten up some weird User Interface things. The Filename is another such crazy. It should be in the window Caption where pretty much ever other app on the planet puts it.

fieldOfView, 2019-11-01:
The problem is that there are other scripts that other users DO want to carry over between projects, eg fixes to gcode files that are necessary for some printers/workflows. And yes, I can even think of workflows where always pausing after the first layer would be required.

>The Filename is another such crazy. It should be in the window Caption where pretty much ever other app on the planet puts it.

There's a plugin for that in the Marketplace: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/VersionInTitlebarPlugin
Pro-tip: do not tack on unrelated issues in an issue queue. That's a sure way not to get those additional issues fixed.

ninovanhooff, 2019-11-04:
We discussed the notification of active postprocessing scripts in the team, and think that the current behavior is preferred for most use cases. 
We might reconsider when a lot of other reports come in about this being unclear. In such a case we will create a new issue more focused on that. Since some of the discussed issues actually have workarounds.

closed 2019-11-04 by ninovanhooff