Ultimaker/Cura #6598: no supports, machine selection related

Docteurg, 2019-10-30:
**Application version**
Cura 4,3,0

**Platform**
Mac OS high Sierra

**Printer**
Kossel linear +

**Reproduction steps**
1. Something you did.
sliced my model for CR10 and printed it
2. Something you did next.
sliced again for my Kossel, impossible to get the software producing supports. tried a lot of different settings

**Actual results**
impossible to print on the cosset, slicing with cura

**Expected results**
(What should happen after the above steps have been followed.)get supports created as with the cr10 profile

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

Docteurg, 2019-10-31:
> Hi @Docteurg <https://github.com/Docteurg>,
>
> How did you add the Kossel (what profiles did you use)? Was it just added as a generic/custom DeltaBot printer?
>
> In any case, it might help if you send us the project file (save, do not export, zip the file to get github to accept it).
>
> cheers,
> Remco.

Hello 

The problem was the "tree support" case checked but not visible on the profile unless unfolded ... that didn't create support at all (which is also an issue)
So : Problem solved for my print

Thanks for your reply

Yes, I added the printer by configuring a generic printer

Here a re the files

Ghostkeeper, 2019-10-31:
Please know that attachments to e-mail replies to Github won't get posted on Github, so if you added files to your e-mail reply we won't be able to see them.

closed 2020-06-19 by no-response[bot]