Ultimaker/Cura #6599: [BUG] Print head keeps moving to the end point

hussainsail2002, 2019-10-30:
**Application version**
Cura 4.3
(The version of the application this issue occurs with.)

**Platform**
Using Windows 10
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)

**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)
Custom Machine

**Reproduction steps**

1. Downloaded Cura 4.3
2. Created a profile for my machine under custom FFF section
3. Loaded model and entered the slicing parameters
4. Saved sliced file on SD card
5. When I hit print the bed moves down by 15 mm and the print head moves to X max and Y max
6. I restart the machine and hit print (no change in GCode)
7. Prints properly without any issue.
8. Maybe design is the problem try another part.
9. First time the same problem occurs
10. Restart machine and print its fine

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
Was expecting the machine to print well from the first time instead of having to restart the machine.
(What should happen after the above steps have been followed.)

**Additional information**

Added the sliced Gcode files.

Thank you
Hussain

[CFFFP_40mm_cube.zip](https://github.com/Ultimaker/Cura/files/3788817/CFFFP_40mm_cube.zip)

(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

hussainsail2002, 2019-10-30:
Hi Remco, 

If there is an issue with start g code, how come it prints correctly on the second attempt after I kill the power and try printing again ?. 

I will nevertheless try and make changes in start g code. I am using the default start gcode that cura comes with.

Ghostkeeper, 2019-10-31:
If the same g-code behaves differently after restarting your printer than before restarting, that must be a bug in the firmware. The same g-code is supposed to behave the same every time.

I don't think there is really a problem with your start g-code. But it is as rburema said: The start g-code commands the build plate to move down (or the nozzle up, relatively seen). That's expected. And I think that's a fine command to do because it prevents scratching the build plate very hard if it's not entirely flat.

After that, the nozzle moves diagonally to the positive endstops, you say. In the g-code I see that it's moving to whatever location your firmware thinks is `X119.812 Y119.658`. So apparently that location is outside of your build volume. There are two things you can try:
* Make your build volume smaller, or...
* check the "Origin at Center" checkbox in the machine settings.

Which of these two is the answer depends on your printer. If you command the printer to move to `[0,0]`, does it move to the centre of the build volume?

nallath, 2019-10-31:
I deleted the comment because it was a duplicate.

Block137, 2019-10-31:
What firmware are you running?
If it's Smoothieware, there is a bug with G92 command (if you used that)
G92 X100 will set current X postio

hussainsail2002, 2019-11-01:
@Ghostkeeper Thanks for replying,

I would like to add a little more clarity as I was a bit vague previously.
Basically what happens is that after the bed moves down by 15mm it does not move back and the print head starts moving around. ( I assumed it was moving towards the end positions and I panicked and hit the power off button)

The origin is 0,0 min position, the G28 command works fine also I printed a  large 300 X 300 cube and waited for the outline to finish and it was good.

The reason why I have a feeling that its a slicer issue is because slicing on simplify3D works every time. I use a different start Gcode on it though

@Block137 
I am using marlin firmware V 1.1.X  downloaded it couple of days back. Is it possible that under the G code flavour settings something else needs to be done

Another thing I noticed is that I have brought in my machine profiles from cura 4.2 , when I open cura 4.3 it says those settings are corrupted or something like that. I usually did not bother with it and just cancel the dialog box, maybe I should create a fresh profile for my machine. 

I apologize as I have said previously that I created a "new profile". I will create a new profile again and test and get back to this post.

Thanks,
Hussain

hussainsail2002, 2019-11-01:
so I created a new machine profile and tried again. The issue still persists.

This time I decided to wait till the machine crashed instead of turning the power off. following is the sequence of events that took place

1. Upon heating of bed and nozzle all the 3 axis homed correctly.
2. The bed moved down by 15
3. Head moved to (300,300) position (max pos)
4. The extruder started to run at high speed and started dumping filament out, the drive gear jammed, but continued spinning at a fast rate.
5. I turned off the machine.

Regards,
Hussain

Ghostkeeper, 2019-11-01:
Only part of that is visible in the g-code. I've annotated what we can see in the g-code vs. what happened for you:

```
G28 ;Home ;Working correctly (step 1.)
G1 Z15.0 F6000 ;Move the platform down 15mm ;Working correctly (step 2.)
;Prime the extruder
G92 E0 ;Reset the E position. Nothing visible happens.
G1 F200 E3 ;Extrude 3mm of filament in about a second. This does some priming. You probably missed this.
G92 E0 ;Reset the E position again. Nothing visible happens.
G92 E0 ;Unnecessary resets probably added in the start g-code as well as in CuraEngine for safety.
G92 E0
G1 F1800 E-1 ;Retract 1mm of filament. You probably missed this since it's so small.
;LAYER_COUNT:199
;LAYER:-6
M107 ;Turn off the fans (irrelevant).
G0 F4800 X119.812 Y119.658 Z0.36 ;Move to (120,120) position.
;TYPE:SUPPORT-INTERFACE
G1 F1800 E0 ;Unretract that 1mm from before.
G1 F1350 X120.508 Y119.012 E0.1137 ;Start extruding slowly for the brim.
G1 X120.647 Y118.893 E0.13561
```

I don't see it moving to 300,300 or dumping filament out at high speed. Really, I don't see what Cura could be doing better for you. Did you check your firmware settings? Maybe your steps per mm are off somewhere.

hussainsail2002, 2019-11-01:
Yes I really like  Cura 4.3 cause of its features, but this seems a bit difficult to work with. 

Even I thought I may have made some mistake , but after restarting the machine the exact same Gcode works fine without any re-wiring or changes in firmware. 

Also my other slicers work fine. So I cannot pin point the issue on the firmware yet.

I will next try to slice with Cura 4.2 and check the print

hussainsail2002, 2019-11-02:
Tried using Cura 4.2 , everything is working fine.

I am assuming the issue lies in the Cura 4.3 slicing

fieldOfView, 2019-11-02:
Please inspect the difference between the gcode produced by Cura 4.2.1 and 4.3.

hussainsail2002, 2019-11-04:
Hi,

I am attaching the files for Cura 4.2 sliced files and the cura 4.3 sliced files.

Thanks

[cura_4.3_sliced.zip](https://github.com/Ultimaker/Cura/files/3802938/cura_4.3_sliced.zip)
[cura_4.2_sliced.zip](https://github.com/Ultimaker/Cura/files/3802939/cura_4.2_sliced.zip)

smartavionics, 2019-11-04:
Comparing those two files, the gcode at the start is quite different. In particular, the 4.2 gcode contains lines to select metric mode and use absolute positioning which are not in the 4.3 gcode. Also the homing is different, 4.2 gcode homes x and y first and then z. 4.3 gcode has a single G28 which, presumably, homes all 3 axes together?

hussainsail2002, 2019-11-04:
@smartavionics 
I usually do not change the start Gcode or add anything.

I use what is default. 

A single G28 command also follows the same sequence of doing X,Y and then Z.

I will look into the metric mode and absolute positioning, thanks for pointing that out.

Ghostkeeper, 2019-11-04:
The start g-code of Custom FFF printer is exactly the same between 4.2 and 4.3 by default. It looks like that 4.2 starting g-code is sliced for a different printer or has modified the start g-code.

fieldOfView, 2019-11-04:
The 4.2 gcode is sliced for a printer with a printer definition that does not seem to ship with Cura (3ding_Hydra200), the 4.3 gcode is sliced for a "Custom FFF Printer".

hussainsail2002, 2019-11-04:
@fieldOfView 

Thanks for pointing that out,

I will try copy pasting the start Gcode of 4.2 into 4.3 and trying it.

hussainsail2002, 2019-11-18:
Hi ,

I tried copying the start Gcode from cura 4.2 into cura 4.3 and the Gcode worked correctly.

I am not seeing any issues anymore in cura 4.3 anymore. Though the issue has been solved for me, I suspect that there may still be an issue with the start Gcode in cura 4.3 as I was only using the default. 

Thanks
Hussain

closed 2019-11-18 by hussainsail2002