Ultimaker/Cura #6601: [4.3.0] Starting and ending g-code sections not accessible

dougscheck, 2019-10-30:
**Application version**
Cura 4.3.0

**Platform**
Windows 10 Home build 1903 x64
Intel HD Graphics 620

**Printer**
Creality Ender 3

**Reproduction steps**
1. Uninstalled - Reinstalled Cura
2. Removed other visual c++ redistributables
3. Reinstalled Cura

**Actual results**
When you open machine settings, under printers, the text fields for starting and ending g-code are not accessible, but the extruder text fields are accessible.

**Expected results**
Should be able to edit the starting and ending g-code for the printer.

**Additional information**
[image]
[image]

nallath, 2019-10-31:
The machine dialog is scalable. You can just increase the size by dragging the bottom.

dougscheck, 2019-10-31:
I was trying to scale the g-code dialog box, not the entire machine settings box. I feel a little embarrassed now. Do you know why the settings came in the way they did? With no default start or end g-code?

dougscheck, 2019-10-31:
Never mind, just had to keep scaling up the dialog box and the default g-code is there

Thanks for the help.

closed 2019-10-31 by dougscheck

Ghostkeeper, 2019-10-31:
Duplicate of https://github.com/Ultimaker/Cura/issues/6253

It seems that the scale of that window is dependent on DPI of the screen or somesuch.

dougscheck, 2019-10-31:
No worries, it is a little confusing to not see the little scaling marks in the lower right corner, you would not believe the window can be scaled. But as long as it works, I am good.
Thanks again