Ultimaker/Cura #6602: [4.3] No retractions while printing support

aricblunk, 2019-10-30:
**Application version**
4.3

**Platform**
Windows 10

**Printer**
Monoprice Maker Select v2, using Prusa i3 preset in Cura

**Reproduction steps**
1. Print a model with multiple support structures
2. Observed that retraction worked fine for slices of the object, but was not happening for support structures.

**Actual results**
Large amounts of stringing between support structures, but not parts of the object.

**Expected results**
No stringing between support structures, because the filament should be retracted before moves.

**Additional information**

Note there's no stringing in between the multiple objects being printed, or on the small vertical structures next to each other which are part of the object, but significant stringing in between the support structures.

[image]

[image]

The model being printed: https://www.thingiverse.com/thing:3181028

Configs:

[prusa_i3_pla_profile_1.inst - Copy.cfg.txt](https://github.com/Ultimaker/Cura/files/3790768/prusa_i3_pla_profile_1.inst.-.Copy.cfg.txt)

[prusa_i3_extruder_0_%232_pla_profile_1.inst - Copy.cfg.txt](https://github.com/Ultimaker/Cura/files/3790770/prusa_i3_extruder_0_.232_pla_profile_1.inst.-.Copy.cfg.txt)

smartavionics, 2019-10-30:
Hello @aricblunk , please save the project file (File -> Save), zip it or rename to .zip and attach to this issue. That way, we get to see exactly what the model and settings are together. Thanks.

aricblunk, 2019-10-31:
Cura save:

[PI3_RPi_Display_Case_Top_Vent_GPIO.zip](https://github.com/Ultimaker/Cura/files/3792211/PI3_RPi_Display_Case_Top_Vent_GPIO.zip)

gcode:

[PI3_RPi_Display_Case_Top_Vent_GPIO (2).zip](https://github.com/Ultimaker/Cura/files/3792215/PI3_RPi_Display_Case_Top_Vent_GPIO.2.zip)

smartavionics, 2019-10-31:
Thanks for the files.

Set combing mode to "not in skin" and set max comb distance without retract to 10 (or similar). You will then get retractions for those long travels.

[image]

Ghostkeeper, 2019-10-31:
Better yet, uncheck the setting "Limit Support Retractions". That will make it retract when moving from one piece of support to another but not necessarily when staying within one piece of support.

I think that answers the question. We already have a feature (or two) to configure this. I hope that's sufficient! We'll close the issue but if it's not working correctly we can reopen of course.

closed 2019-10-31 by Ghostkeeper

aricblunk, 2019-10-31:
Thank you both. I would prefer support retractions enabled by default, but it's nice to know it's an option. This can stay closed. Happy halloween!