Ultimaker/Cura #6604: weird support creation

mehran-Bgh, 2019-10-31:
**Application version**
cura v:4.1.0

**Platform**
windows 10

**Printer**
it's a custom printer. it's irrelevant to the issue.

**Reproduction steps**
I just was testing the software and the G-code it generates.
import an STL which was a regular triangle. the stl :
[Part1.stl.zip](https://github.com/Ultimaker/Cura/files/3792788/Part1.stl.zip)

check the parameters (the generate support structure was activated)

**Actual results**
I check the preview and find out that a weird support structure has been generated.

[image]

I check the G-code file as well and some unnecessary G-codes for support were there. 

> ;TYPE:SUPPORT
> G1 F1500 E177.50627
> G1 F1800 X108.355 Y94.777 E177.52269
> G1 F1500 E171.02269

here is the 3mf file
[CFFFP_Part1.3mf.zip](https://github.com/Ultimaker/Cura/files/3792781/CFFFP_Part1.3mf.zip)

**Expected results**
this model didn't need any support structure but it was generated.

**Additional information**
however, this issue will solve if we uncheck the support generating checkbox but creating unnecessary support while generating support is activated is insane.
[image]

Ghostkeeper, 2019-10-31:
Yup, can reproduce with the default settings of Custom FFF Printer (making it 200x200mm wide so that this model fits in it by default). The model is really simple, just 6 vertices and 8 triangles.

mehran-Bgh, 2019-11-01:
what's the reason then? does it seem rational to you?

Ghostkeeper, 2019-11-04:
No, it's very weird and shouldn't be there. I have no idea why it chose to create support for that particular corner. Definitely a bug.

nallath, 2019-11-04:
For our reference CURA-6955

yohanson, 2019-11-12:
Reproduces in 4.4.0-BETA.
Weird support generation depends on overhang angle setting.
[image]
Can be somewhat worked around by setting 50° or more.

3DTanny, 2019-11-15:
Using 4.3.0, getting random bits of support lines being generated in mid air, supports not reaching the overhang they are intended to support, and even support lines crossing over perimeter lines (although that has only happened once and I wasn't able to produce again after a re-slice.)

If any screenshots/profiles/gcode files would help, I can provide them.

Ghostkeeper, 2019-12-03:
As far as I'm concerned this ticket is not about "there is support in weird places", which is a very broad topic concerning dozens of algorithms and thousands of lines of code. The problem here is only concerned with the case of support being generated alongside a perfectly vertical wall. So I'll change the title to prevent this issue from becoming a collection of all support issues out there.

Ghostkeeper, 2020-04-21:
This was fixed by Ultimaker/CuraEngine#1214, which will be included with Cura 4.7.

closed 2020-04-21 by Ghostkeeper

marhub, 2021-02-26:
This issue still persists in cura 4.8.0

[image]

Ghostkeeper, 2021-03-04:
@marhub There is a similar issue related to support towers being generated but then being removed due to X/Y distance, which is what you're seeing there. Please refer to https://github.com/Ultimaker/Cura/issues/7487.