Ultimaker/Cura #6611: usable documentation for "Running Cura from Source on Ubuntu"

ortylp, 2019-10-31:
Unfortunately the package for Cura in Ubuntu is broken, so I had to compile it from source.

I have used instructions from
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu

It mostly worked, but copying commands line by line from wiki is to say the least... infuriating.

As a result I have created a script that gets the sources and builds the thing.  The following commands have been tested with a vanilla installation of ubuntu-19.10-live-server-amd64 (on 31.10.2019).

Please make this or similar script available from the wiki web page.  A proper (build) shell script would avoid copy&paste code, but the version below is easier to inspect and shows the similarity to the instructions on the wiki page.

```
#!/bin/bash -x

# required for build:
sudo apt install build-essential cmake git python3-pip python3-scipy python3-sip-dev zlib1g-dev libssl1.0*-dev
pip3 install pyserial PyQt5==5.10 shapely zeroconf trimesh

# required for execution:
sudo apt install libgl1-mesa-glx libfontconfig1 libxrender1

# build Cura in ~/tmp/cura

mkdir -p ~/tmp/cura && pushd ~/tmp/cura

git clone https://github.com/google/protobuf.git
pushd protobuf
git checkout 3.5.x
mkdir build && cd build
cmake ../cmake -Dprotobuf_BUILD_TESTS=OFF -DCMAKE_CXX_FLAGS=-fPIC
make -j 8
sudo make install
popd

git clone https://github.com/Ultimaker/libArcus.git
pushd libArcus
mkdir build && cd build
cmake ..
make -j 8
sudo make install
popd

git clone https://github.com/Ultimaker/libSavitar.git
pushd libSavitar
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$VIRTUAL_ENV -DCMAKE_PREFIX_PATH=$VIRTUAL_ENV -DBUILD_STATIC=ON -DBUILD_PYTHON=ON -DBUILD_EXAMPLES=OFF ..
make -j 8
sudo make install
popd

git clone https://github.com/Ultimaker/libCharon.git
pushd libCharon
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$VIRTUAL_ENV -DCMAKE_PREFIX_PATH=$VIRTUAL_ENV ..
make -j 8
pip3 install ..
popd

git clone https://github.com/Ultimaker/CuraEngine.git
pushd CuraEngine
mkdir build && cd build
cmake ..
make -j 8
sudo make install
popd

git clone https://github.com/Ultimaker/Cura.git
git clone https://github.com/Ultimaker/Uranium.git
git clone https://github.com/Ultimaker/fdm_materials.git

cd Cura
ln -s /usr/local/bin/CuraEngine .
mkdir resources/materials
ln -sf ../../../fdm_materials resources/materials/fdm_materials

# to start the application use the following command:
# PYTHONPATH=~/tmp/cura/Uranium:${PYTHONPATH} python3 cura_app.py
```

Ghostkeeper, 2019-11-01:
That's useful, but we've also created one. Or actually, two; one for getting the dependencies and another for getting the main sources:
* https://github.com/Ultimaker/cura-build-environment/blob/master/CMakeLists.txt gets the dependencies.
* https://github.com/Ultimaker/cura-build/blob/master/CMakeLists.txt gets the main source. This is not only used to allow building an installer but also checks out prefixes for all of these repositories that can be used to run from source.

These are also cross-platform. We've tested them on Windows, MacOS and Linux, and several distributions of each of these. And it avoids copy/pasting code from a website into your terminal, which is generally bad practice.

rburema, 2019-11-01:
In addition to that:
 > Unfortunately the package for Cura in Ubuntu is broken, so I had to compile it from source.

You can also download the AppImage (via our site or more directly from) [here](https://github.com/Ultimaker/Cura/releases/download/4.3.0/Ultimaker_Cura-4.3.0.AppImage). It should work on most linux distros.

ortylp, 2019-11-01:
@Ghostkeeper I like this solution. The follow-up question would be: why is it not described on the "build from source" wiki page.  I assume it would require one git clone and two cmake calls.  Please make it happen :D

@rburema I was looking for a snap package first.  Bonus: I learned about alternatives to snaps, but I would still prefer a snap with sandboxing...  Especially, that snaps can automatically pull for updates.

For now I am set, with a script to build and an alternative to pull and rebuild.

I just wanted make the documentation here more usable for desperate newcomers needing a build from source.  Maybe a hint about a AppImage (and hopefully a snap in near future) could also be included.

Ghostkeeper, 2019-11-04:
I've written down the expected workflow for CMake in the Wiki now.

The AppImage is our main build file for Linux. I'm supposing that if you're looking to build Cura from source, you've already had experience with the build that we supply or know that it exists in Github's releases page.

closed 2019-11-04 by Ghostkeeper