Ultimaker/Cura #6612: Cura 4.3 - Infill layer thickness greater than layer height causes infill not to fuse correctly to the shell

johnkray, 2019-10-31:
**4.3**

**Windows 10 (this issue is not system dependent)**

**Hydra Research Nautilus (this issue is not printer dependent)**

**Reproduction steps**
Change the infill layer thickness to a multiple of the layer height i.e. layer height = 0.1, infill layer thickness = 0.2.

**Actual results**
There is a gap between the infill and the shell.

**Expected results**
The infill should fuse with the shell as specified by the infill overlap percentage. 

**Additional information**
I'm assuming that this issue occurs because Cura treats an infill layer thickness that is thicker than the base layer height as a wider extrusion of the base layer height. I can elaborate more here if needed.

smartavionics, 2019-10-31:
I think a fix for this (https://github.com/Ultimaker/CuraEngine/pull/1003) was recently merged into Cura and should be in the next release.

johnkray, 2019-10-31:
Great, thanks. I will wait for the next release and report back if there are any issues.

Ghostkeeper, 2019-11-01:
Yes, that should've been the fix for this issue.

closed 2019-11-01 by Ghostkeeper