Ultimaker/Cura #6613: Heat creep during ironing, suggest a large retract before ironing large areas.

iburnthings, 2019-10-31:
I just tested out the ironing feature on a lithopane that had a large 'base' area. Since I'm printing flat I wanted this area smooth. 

During the ironing I have 10% flow which seemed too much but eventually my extruder started clicking from a clog. I pulled the filament out and it was heat creep. I thought maybe if there was a way to retract like 100mm of filament before ironing a large area it would prevent that.
CR10S, eSun white PLA+ 190c nozzle, 55c on the bed.
Please let me know if you think this is a good idea or a bad one.

I could also turn down the flow for ironing but then I think more heat creep will occur.
I could turn down the heat but then it might not iron it properly and it might take too long to get up to temp again to start printing again.

iburnthings, 2019-11-01:
I just realized a separate ironing temperature might work though since it doesn't necessarily need to be as hot as a printing temp

Ghostkeeper, 2019-11-01:
The effectiveness of ironing is not really due to just stroking over it with a hot piece of metal. It's more that it's filling the crevices with new material. Retracting the filament would defeat that purpose. You won't have the 10% flow any more.

We don't have temperatures per structure since the PID regulators in many printers don't handle frequent changes in temperature very well and start to fluctuate the temperature wildly. In general if the temperature is changed for less than a minute you probably shouldn't change the temperature.

Maybe it's worth some experimenting if a "dry run" with retracted filament is actually effective for you. You can do this by setting the flow to 0% and then editing the g-code to put a retraction just before it starts ironing. If you're using absolute E values to retract, it'll automatically unretract during the first line after ironing then.