Ultimaker/Cura #6619: [4.3.0] Problem resizing without uniform scaling

winijanc, 2019-11-03:
When scaling without uniform scaling, the percentages and values ​​of the X, Y, and Z axes become messy, depending on which information changes, another value also changes incorrectly. It looks like an interface problem.

[image]
[image]

Kingmiwok, 2019-11-03:
I'm having a similar problem, but it seems to be tied to rotating the model.  Non-uniform scaling works fine unit I rotate the model, then afterward scaling gets weird, and the axis's get jumbled.  It seems like if you rotate an object on an axis, then scaling association should move with the axis, but it does not.  In my case, if I scale first, then rotate, it's fine (except the scale percentage is associated with the wrong axis).  But if I rotate first, and then try to scale, I have the problems like described above.
This works.
[image]
[image]
[image]

Changing the order (rotating first, and then scaling) does not.
[image]
[image]
[image]

Ghostkeeper, 2019-11-04:
Indeed, rotating the model + non-uniform scaling causes this issue. It's a very difficult one.

This is a duplicate of https://github.com/Ultimaker/Cura/issues/4136. Please continue the discussion there.

closed 2019-11-04 by Ghostkeeper