Ultimaker/Cura #6621: Bug form

userosos, 2019-11-03:
Cura 4.3
I want make an form for silicone bit if i use the form the first layers are hanging in the air.
I want that the first layers aren't hanging in the air.
3mf file:
https://we.tl/t-s8TcXOwXvC

nallath, 2019-11-04:
I don't understand what you mean. What is it exactly that you're trying to do (and failing to do?)

userosos, 2019-11-04:
@nallath 
I user Mold and it work not ok.
[image]
[image]
[image]

nallath, 2019-11-04:
Did you check if your model is watertight? I've just tried it and it works for the models I used.

Ghostkeeper, 2019-11-04:
The model seems to be watertight. The problem, I think, is an expectation problem. The user's model actually contains a base plate (like the bottom of a box). You can see it in X-ray view as the dark underside of the model, where the X-rays go through only 2 faces.

[image]

If you were to slice normally (without mould enabled), this produces a base plate of 5 layers thick. With mould enabled, it produces a hollow area of 5 layers thick (using your 3MF file's settings, which are apparently different from the 3 layers you reported). This is intended because Cura thinks you want to cast the base plate as well. It doesn't know that you only want to cast those 3 protruding parts of your model.

Ghostkeeper, 2019-11-04:
If you move the model 1mm down it'll behave as you'd expect because Cura won't create a mould for the piece that is below the build plate (i.e. the base plate of your model).

We'll close this because it's expected behaviour.

closed 2019-11-04 by Ghostkeeper