Ultimaker/Cura #6626: Cura 4.3 Network enabled printer keeps changing

V3DPrinting, 2019-11-05:
**Application version**
4.3.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)
macOS 10.14.6

**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)
Ultimaker S5 was connected as a networked printer, host of Cura Connect with 2 Ultimaker 3 extended 

**Reproduction steps**
1. Something you did.
Connected the UMS5 to Cura as host

[image]

2. Something you did next.

I then changed the printer in the dropdown list to the UM3E

**Actual results**
(What happens after the above steps have been followed.)
This has changed the networked printer to UM3E on top of changing the selected printer
As a result the printer connected was not allowed, as not being the Cura Connect group host.
It has also created a new connected printer with the UM3E

[image]

**Expected results**
(What should happen after the above steps have been followed.)
Normally, only the selected printer would change. The connected printer should stay the Cura Connect group host, so to say the Ultimaker S5

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

[cura copie.log.zip](https://github.com/Ultimaker/Cura/files/3809142/cura.copie.log.zip)

Ghostkeeper, 2019-11-06:
I'm unable to reproduce this problem. These were my reproduction steps:
1. Connect an Ultimaker 3 Extended printer (called DeWey, for reference) as a slave to an Ultimaker S5 printer (called Lily). Lily is the host.
2. Add a printer in Cura.
3. Select Lily from the discovered printers list. DeWey is not in that list since it's not a print cluster any more. It is listed in grey at the bottom of the list.
4. In the printer configuration drop-down, change the printer model to Ultimaker 3 Extended.

The results I'm observing are:
* The printer in Cura is still called Lily.
* When sending the print I can choose either printer or automatic. If I choose the Ultimaker S5 then it'll say that the configuration doesn't match, indefinitely.
* When sending the print using the automatic selection or the UM3E then it'll start a print properly.

Those are expected results. Am I missing something?

V3DPrinting, 2019-11-06:
> * When sending the print I can choose either printer or automatic. If I choose the Ultimaker S5 then it'll say that the configuration doesn't match, indefinitely.
> * When sending the print using the [...]

If you are able to select the printer you want to print out, then I suspect you are not using Cura 4.3.0, as this feature has been disabled by Cura Team in that release.

The bug I have reported never showed up in 4.2 or previous versions.

Ghostkeeper, 2019-11-06:
Yeah I'm using the 4.4 beta. We added it back in!

V3DPrinting, 2019-11-06:
Good news for the printer selection feature !

I've attached some more information, as my bug is still there.
This morning I've started Cura and connected automatically on the S5 Networked printer called Ultimaker-007483. No changes from yesterday's configuration.

Right now it isn't any more connected and I have new supposedly connected printers created in my printers' list.
But in fact Cura is not anymore connected to the group host.

[image]
[image]
[cura.log.zip](https://github.com/Ultimaker/Cura/files/3814154/cura.log.zip)

Before going to a beta version, I prefer downgrade to a stable version if possible.
I'm a 3D Printing service provider and it's my business, so I prefer to play it safe.

Ghostkeeper, 2019-11-07:
Then you'd have to wait a few weeks for the stable version. Or try it on a different computer or a virtual machine.

We haven't seen the problem here yet. From your logs, it doesn't seem to be related to cloud-connected printers. You could try temporarily uninstalling the sidebar plug-in in the off chance that it's just an interface issue.

V3DPrinting, 2019-11-08:
I have done a fresh install of Cura 4.3.0 on my Mac (10.14.6) and restored my backup.
But I haven't uninstalled the Sidebar plugin.
I still have the same issue, but noticed it appears when loading a project which is using an UM3E.
If the project is using the UMS5, no networked printer change.
It also has created a UM3E networked printer again.

Maybe it helps.

nallath, 2019-11-11:
It could also be that the issue is caused by loading a project. Could you share that project file?

V3DPrinting, 2019-11-11:
I have the issue whatever the project file is, for the time it's based on the non group host printer.
I only work by project, so it is easy to retrieve all the settings for a specific part / customer order.

I cannot share the projects which created the issue for confidentiality reasons, but as soon as I'm in the office I will try to reproduce the problem with a dummy project and share it to you.

nallath, 2019-11-11:
It's mostly the steps that I'm interested in. So if you can find a set of steps that reproduces it with a newly created printer, that would be best (Preferably also with the project being newly created, so we can make our own projects that have the issue)

V3DPrinting, 2019-11-11:
Nallath

I've done the following
Reset the Cura Connect Group.
Cura Backup
Trashed the Cura App and the User/Library/Application Support/Cura folder
Fresh install of Cura 4.3.0 on my macOS 10,14.6.
Added the Cura Connect Host (UM S5) manually by IP
Connection to Ultimaker Cloud + Sign in
Restored the last backup and restarted Cura
Checked all the plugins, profiles and materials
Cleaned all the remaining Networked printers from last install
[image]

So before starting any project I have the following configuration regarding the printers
[image]

I've created a dummy project on the UM S5 with my profiles and materials.
[image]

Here's the project file
[UMS5_LFS_Elephant.3mf.zip](https://github.com/Ultimaker/Cura/files/3832065/UMS5_LFS_Elephant.3mf.zip)

Then switched the printer to UM3E, still having the UMS5 as networked printer selected
Created a new project with my profiles and materials and the same model
[image]

Here's the project file on the UM3E
[UM3E_LFS_Elephant.3mf.zip](https://github.com/Ultimaker/Cura/files/3832083/UM3E_LFS_Elephant.3mf.zip)

Then opened the UM S5 project
[image]
[image]

The networked printer is apparently still the UM S5, but it has created a new networked printer with the same name. The other one is not connected.
[image]

I can still print other network on that printer.

Then I opened the UM3E project updating the printer, profile and material
[image]
[image]

The UM S5 is still selected as a networked printer and I can still print over network.

Opening again the original UM S5 project leads to an issue where the material is not anymore the one selected in the project
[image]

And it has created a new instance of the UM S5 as a networked printer
[image]

I've reopened the UM3E project with no issues.

I've opened the UM S5 then UM3E project a couple of times without being able to have the UM3E connected as a networked printer (which has happened a couple of time last week).
Sometimes I got the material changed to a different one than in the project
Each time I've loaded the UM S5 project, I had a new networked printer added.
I also tried with a project from last week, but without managing to have the UM3E as a networked printer.

Here's the log file
[cura.log.zip](https://github.com/Ultimaker/Cura/files/3832256/cura.log.zip)

Definitely there is something wrong with the projects (networked printer created though already there and sometimes material changed) but I haven't managed to reproduce the error.

Hope it helps !

nallath, 2019-11-12:
I think there is something in your local config that is causing it. So it might actually be the backup causing the issue to occur (since that puts the exact same printer back again). 
Could you try and see if you can reproduce the issue if you add a new printer and connect that one with the network one?

V3DPrinting, 2019-11-13:
Thanks Nallath

I do confirm that the problem occurs when opening a project were the printer is updated.
I am currently preparing to make a fresh install the old (and hard) way, exporting all the profiles and material before deleting the app and application support files.

V3DPrinting, 2019-11-13:
Nallath

I have made a fresh install of Cura 4.3.0 :
Trashed the Cura App and the User/Library/Application Support/Cura folder
Fresh install of Cura 4.3.0 on my macOS 10,14.6.
Added the Cura Connect Host (UM S5) manually by IP
Connection to Ultimaker Cloud + Sign in
Ultimaker S5 was connected as networked printer, as the group host
Added local printer for the UM3E
Doing that, connection was lost to the networked printer and I had to reconnect to the network printer.
No plugins, no profiles and no materials added
Created a project on the S5 using default fine profile and Ultimaker PLA Black
[UMS5_LFS_Elephant.3mf.zip](https://github.com/Ultimaker/Cura/files/3840628/UMS5_LFS_Elephant.3mf.zip)

Switched to UM3E
Created a project using the default fine profile and Ultimaker PLA White
[UM3E_LFS_Elephant.3mf.zip](https://github.com/Ultimaker/Cura/files/3840638/UM3E_LFS_Elephant.3mf.zip)

Opened the UMS5 project then the UM3E project

And the bug appears !
[image]

A new networked printer has been created when opening the projects.
It's a problem of the 4.3 release, not the backups.

Here's the Cura 4.3 folder within /User/Application Support
[4.3.zip](https://github.com/Ultimaker/Cura/files/3840679/4.3.zip)

V3DPrinting, 2019-11-13:
I will revert to 4.2.1, even if I loose some settings.
The bug is too impacting for my business as a service provider, were I constantly switch from one project to another on both types of printers.

nallath, 2019-11-13:
I've just tried to do this with 4.4 (it's in beta now) and I can't reproduce it. I do see that some changes were made in some bits of code that affect this.

Could I ask you to check that one final time?

V3DPrinting, 2019-11-23:
Hi Nallath
I've migrated to 4.4 and I do confirm I still have the bug.
Projects are useless with that bug.

V3DPrinting, 2019-11-24:
Hi have noticed that the networked printer (UM S5 as the group host) keeps disconnecting, most of the time when performing a intensive compute, but also when Cura is idle.

Ghostkeeper, 2019-12-03:
Maybe it depends on the specifics of the network, i.e. disconnecting momentarily for certain amounts of time?

V3DPrinting, 2019-12-03:
I have reset all the network devices in my local network to see if it changes something.
The disconnections are still there. In fact it disconnects then reconnects right away to the networked printer without changing if there is no compute on Cura at that time.

If there is an activity on Cura, then it disconnects without reconnecting.

But if I load a project, there is a fair amount of chance to be disconnected, especially if the selected printer is not the same as in the project.

My feeling is project is corrupted on save due to the disconnection, then loading it leads to set up a new printer, not connected.

The issue is on 4.3 and 4.4.

Ghostkeeper, 2019-12-03:
If the project has anything to do with the network connection, then I'd expect that saving the project with an empty build plate will still reproduce the problem. The project can affect the network connection since the metadata of the printer is stored in the project, which includes the printer's Zeroconf address. So if the models you're working with are confidential, you may still be able to share the project when saved with an empty build plate :smiley:

V3DPrinting, 2019-12-03:
@Ghostkeeper 

Sorry, but I have downgraded to 4.2.1 this morning. Fresh install, only imported the materials and profiles the old way, not from a backup.

Which hasn't disconnected from my network since.... So there is something wrong in 4.3 and 4.4.

But you can retrieve one of the projects posted above in the thread UMS5_LFS_Elephant.3mf.zip and UM3E_LFS_Elephant.3mf.zip.

Ghostkeeper, 2019-12-05:
We've had a refactor of the UM3NetworkPrinting plug-in in Cura 4.3 that seems to have thrown some things in disarray. Perhaps @ChrisTerBeke has more of an idea as to how we can reproduce this issue on our side?

ChrisTerBeke, 2019-12-05:
I think this was recently fixed by @nallath in https://github.com/Ultimaker/Cura/commit/45bbb189ce8bbae173740e22ddf392fe296ad02b? So in that case it'll be fixed in 4.5.

V3DPrinting, 2019-12-05:
To give my two cents, since I'm back on 4.2.1 I don't have anymore disconnects nor new machines creations.
No more network issues, though I've changed nothing in my network configuration nor computer.

Definitely the issue is in Cura since 4.3

closed 2020-06-19 by no-response[bot]