Ultimaker/Cura #6627: Print Sequence One at a time - Last model is not locked to height limit

Quasardian, 2019-11-05:
**My Problem:**
Not related to a bug, but on a few occasions I've tried to use the feature print one at a time with multiple objects.
Every time I'm locked to the height limit when only the last object is too tall, while this shouldn't matter then.

**Solution:**
The tallest model outside of height limit can still be printed.

**Alternatively:**
I also thought about printing multiple object one at a time which are outside of the height limit.
Then to avoid the printed parts, axis, and travel towards the head switcher the print order should be from the bottom left to top right.

**Current printers:**
S5 (3x)

Ghostkeeper, 2019-11-07:
We get this request once in a while. Last time was [here](https://github.com/Ultimaker/Cura/issues/4835) but as an aside in a different ticket. One of the objects could in theory be taller than the gantry.  Maybe we can model this by not taking the intersection with the mirrored shape of the print head for models taller than the gantry height, and then still only allowing one model that is taller.

rburema, 2019-11-07:
We've discussed this and chosen to defer it for now.

Sophist-UK, 2020-04-09:
IMO Cura should be a little more intelligent and automatically place the single object that is taller than the height limit as the last object to be printed.

I think that coding this should be within my Python skills - I will try to find the time to submit a PR for it in the next few weeks.

Sophist-UK, 2020-04-14:
I have started looking at coding this and there are multiple possible solutions:

a. We can allow a single part greater than the gantry height and print that last.

b. Rather than reducing the build volume (which feels odd to me), we can instead extend the existing clash checks for head hitting other objects as follows:

* for objects lower than the gantry height, we only need to check for head clearance as currently done with "reduced build volume"; 

* for objects greater than the gantry height, we check for clearance using the same mechanism as present and using the same print head depth and height, but using the full bed width rather than the head width (to account for gantries rather than the head itself). This would allow (as the simplest example) a tall object at the front left of the bed and another tall object at the back right of the bed, with the object at the front printed first, and the object at the back printed second - which is OK because the gantry will remain behind the first object for the whole of the print).  

Cura would need to print objects in the following order:

1. First, those objects less than gantry height in any sequence;
2. Then, those objects greater than gantry height, printed from front to back.

What do people think of this idea?

**P.S. I am having great difficulty in creating a dev env. i.e. running cura from source. Anyone help with this will be gratefully received!!!!**

Sophist-UK, 2020-04-14:
Just to clarify - for objects greater than the gantry height we would change the shape of the shadow (representing the head) so that the area in front of the head point was full width like this. 

[image]

In this graphic all objects are under gantry height, but if you imagine that they were over gantry height I hope you can see that if you printed the front object first, then when you printed the middle object the gantry would not hit the first object, and similarly for the last object.

Sophist-UK, 2020-04-14:
On reflection, I think that the gantry shadow would need to be a separate shadow (mesh) from the head shadow. This is because you the head needs to avoid the below gantry objects and both the head and gantry need to avoid the above gantry objects. Indeed, it might be better to show both of these additional meshes in order to assist the user in positioning objects.

(As a further aside, I personally find the way Cura moves clashing items out of the build volume to be very annoying. I would much rather that only objects that physically clash are moved (in order to avoid the user finding it difficult to select an object) and that the objects which do not clash physically but whose shadows clash are simply marked as unprintable so that the user can tweak their position on the bed.

Sophist-UK, 2020-04-14:
#4835 suggests that we need to decide whether print sequence should be front to back or back to front by virtue of whether the hotend shape has more bulk in front of or behind the nozzle.

It would appear my proposed solution might resolve both this issue and #4835.

rburema, 2020-04-16:
> P.S. I am having great difficulty in creating a dev env. i.e. running cura from source. Anyone help with this will be gratefully received!!!!

I'm off work for the next few days, but [this part of the wiki](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) should be enlightening.

Sophist-UK, 2020-04-16:
I have followed the Windows Wiki entry to the letter but it simply doesn't work for me.

fieldOfView, 2020-04-17:
> it simply doesn't work for me

Perhaps if you are more specific we can help. But probably best to do that in a different thread/issue.
Personally I don't build Cura, but use some precompiled parts from an official version (most notably CuraEngine.exe and libArcus.pyd) and run the Python sources. See [this forum post](https://community.ultimaker.com/topic/29944-how-to-debug-plugins-during-development/?do=findComment&comment=249369) on how I set that up.

Sophist-UK, 2020-04-17:
I have already asked in Community to no avail. In essence, CMake for libArcus doesn't work despite following the instructions in the Wiki to the letter.

fieldOfView, 2020-04-17:
I am not telling you to ask in the community, I am linking to a community post outlining an alternative to building CuraEngine and libArcus. And no matter how many times you say "doesn't work", we will not be able to help you unless you say what error messages you get.

Sophist-UK, 2020-04-17:
Yes - after I posted the above I went to have a look. I will try to give it a go tomorrow - because I only need to run Cura UI from source - the rest can run from executables. Also, I intended to add a link to my Community post which gives full details of all error messages. (As you say, this is not the place for such details.)

Sophist-UK, 2021-01-26:
As stated in a different issue, gantries can vary widely can be x-direction or y-direction or both or indeed neither and are not modelled (yet?) in the Cura printer definitions.

However, gantries are generally in line-ish with the print head, so if we assume both x- and y- gantries, we should be able to position items in a diagonal line e.g. front-left to back-right and use the head mesh for avoidance.

fieldOfView, 2021-01-26:
> However, gantries are generally in line-ish with the print head

Do you know SCARA printers?

Another complicating factor are toolchangers and similar techniques (such as the switch mechanism in use by Ultimaker printers) that can move all over the bed causing possible collisions.

Sophist-UK, 2021-01-26:
No - I don't know scara printers - are they the ones where the head is moved by 3 levers?

Nor have I researched switch mechanisms.

This whole area is complicated.

SMack40, 2021-01-27:
I don't know if this was pure luck, but i had this problem yesterday and for me it seemed that the y-axis is inverted:

Looking at the cura buildplate, you have the origin of the x-y-z-axis' on the front left corner. Positive directions being to the right (x), to the back (y) and upwards (z). 
But if you load an object, x and z remain the same while the positive y direction inverts to the front.

So i took these Object-Koordinate system to describe my printhead and it worked.

Ghostkeeper, 2021-02-04:
It's indeed a complex topic because it's highly dependent on how printers around the world operate, and there are so many of them.

@SMack40 it was sort of luck. The print order is determined in such a way that you can place objects as close as possible, such that the print head could approach from either side, and the print order is then calculated so that the print head will approach from the correct side. This should make the actual collision area around the shape symmetrical on the X and Y axes. However the shadow is currently not calculated correctly as per a known bug. It's too large, which sometimes prevents you from placing objects closer.

romerohe, 2021-02-16:
I've found a temporary solution changing the gantry height parameter for the printer under Preferences/Printer tab, setting a higher value than the largest piece. First piece loaded into Cura was the last printed, and I put the higher one on the top right-corner (home position in my printer is bottom-left).

closed 2023-07-13 by github-actions[bot]