Ultimaker/Cura #6628: Disabled Prime Tower Prevents Slicing Based on Virtual Overlap

AbeFM, 2019-11-05:
**Application version**
Current

**Platform**
NA - win10

**Printer**
Any Dual Extruder

**Reproduction steps**
Enable Prime Tower
Set position where model overlap may occour
Disable Prime Tower
Load model that overlaps with previous Prime Tower Location

**Actual results**
Slicer won't slice, says invalid Prime Tower Location

**Expected results**
Prime Tower should be ignored

**Additional information**
I feel the check for overlap with the prime tower should be skipped when the Prime Tower is disabled.
https://github.com/Ultimaker/Cura/issues/6390#issuecomment-534916593

AbeFM, 2019-11-05:
In trying to come up with graphics for this post, I found I was sometimes able to get my items to slice, so I might have to close this until I can more definitely demonstrate when it is a problem and when it is not.

closed 2019-11-05 by AbeFM

Ghostkeeper, 2019-11-07:
Tried this once (so N=1 for sample size I guess) and it sliced fine. If you find out in what cases it doesn't slice, we'd be glad to hear!