Ultimaker/Cura #6629: Allow Prime Tower within Object if no interference with part

AbeFM, 2019-11-05:
**Is your feature request related to a problem? Please describe.**
Objects, or groups, have an outline that overlaps with Prime Tower, but actual parts clear Prime Tower

**Describe the solution you'd like**
Prime Tower should print in hollow areas if it does not interfere with part.

**Describe alternatives you've considered**
Complaining and hoping people smarter and more beautiful than me will solve all my problems.
Sometimes you can break up your object or ungroup it in such a way as to "unclaim" some empty space and allow the print to proceed, but it's not generally applicable.

**Affected users and/or printers**
Anyone using multiple extruders/materials.

**Additional context**
[image]

By way of example, I have this test for nozzle alignment. To the human observer, it's obvious that the prime tower could sit dead center on the bed.

I recall a moderator mentioning checking for interference with the Prime Tower is non-trivial, so perhaps this is too big an ask.

rburema, 2019-11-07:
The engine is probably almost already able to do it, but it might indeed require _a lot_ of work in the front-end. Also since the more complex collisions require more capacity the efficiency of the interface might go down.

Ghostkeeper, 2019-11-07:
We're currently checking for collisions with the convex hull. This is useful for several reasons:
* It guarantees that travel moves within the part don't hit anything, e.g. when using one-at-a-time mode. This is not applicable to the prime tower (since one-at-a-time mode doesn't work with the prime tower) but to the general problem of using the convex hull to check for collisions.
* A lot of geometric algorithms are faster with convex hulls. These need to be performed in Python in the front end so it's performance-sensitive.
* The convex hull has fewer vertices so everything performs better.

We've chosen to defer this issue since it's theoretically feasible but technically difficult. The engine will blindly slice causing overextrusions, except we subtract the prime tower's area from support to prevent conical and tree support from hitting it.

closed 2023-07-13 by github-actions[bot]