Ultimaker/Cura #6630: Prime tower decreases size of printable area

upatamby, 2019-11-06:
**Cura 4.3.0**

**Windows 10 Pro**

**Ultimaker 3 Extended**

**Reproduction steps**
I have Ultimaker Nylon in extruder-1 and Ultimaker PVA for support in extruder-2.
I load an object with an X-size of 180,2mm.

**Actual results**
If the prime tower is disabled then Cura is able to slice.
But if I enable the prime tower then, for some unknown reason, the printable area becomes smaller and Cura cannot slice the object anymore.

**Expected results**
I don't see why the max-size of the printable area has to be smaller if a prime tower is used. I would expect that a prime tower has no influence on the size of the printable area at all.
[image]
[image]

rburema, 2019-11-07:
This could happen if adhesion is set to none (or not brim) and the prime tower brim is on. Could you check if that is the case?

upatamby, 2019-11-07:
Thanks Remco.
Yes, indeed, adhesion was set to skirt (non-brim) and the Prime Tower Brim was on.

Now I understand how this is build, but I'm not happy with this 'easy' solution, because it causes unnessecary limitations for all the other objects.

As the prime tower isn't a normal object anyway (e.g. you can't move it with the cursor like other objects) I propose the following changes:
- let Cura ignore the brim of the prime tower when estimating the maximum build size
- let Cura handle the prime tower position differently: adjust the default X and Y values automatically and make invalid manual entries red
Cura already does these things, but the calculations must change.

rburema, 2019-11-07:
So this has become a feature request then. I'll discuss it with the team.

Vandrasc, 2020-07-21:
Fixed in CURA-6980, the fix will be included in the next release.

closed 2020-07-21 by Vandrasc