Ultimaker/Cura #6631: Feature request: Settings | Collapse All button

tlhintoq, 2019-11-06:
As referenced on the Cura Forum:
https://community.ultimaker.com/topic/26462-feature-request-settings-panel-collapse-all-button/

Right now every job sliced is the same process:

- Close all the groups
- Open the top Group (Quality) confirm or change
- Close that group, open the next.
- {...} repeat all the way down the settings, times about 15 settings groups.

This is truly tedious job after job.  It would be really helpful if there was a button that would 'collapse all' of the settings groups (Quality, Shell, Infil, Material {...}). So the user can then just walk down the groups one at a time double checking everything, without having to either open/close every group every time or leaving them all open for a massively long list to scroll every time. 
[image]

alexjx, 2019-11-07:
I would like this feature.

nallath, 2019-11-11:
For our reference: CURA-6959

nallath, 2020-01-02:
We just merged this, so it will be in 4.5>

closed 2020-01-02 by nallath

tlhintoq, 2020-03-15:
AWESOME!

LouiseTGUK, 2021-09-05:
Even better.. Have ability to only open one section at a time, so when you open a second one it closed the first..

Many progs with long categorized lists like this have that ability including my photo editor, makes things MUCH easier to navigate