Ultimaker/Cura #6632: Hot end travel to prime tower location even if prime tower option is unchecked

clelong, 2019-11-06:
**Application version**
[G_Phil_Color_2.zip](https://github.com/Ultimaker/Cura/files/3815677/G_Phil_Color_2.zip)

CURA 4.3.0

**Platform**
Windows 10 (64bits)

**Printer**
A20M

**Reproduction steps**
Does it for all the stl files i want to print.
Load stl, preview it and can see that the hot end always moves to the prime tower location even if prime tower option is unchecked.

**Actual results**
The hot end always moves to the prime tower location even if prime tower option is unchecked.

**Expected results**
(What should happen after the above steps have been followed.)

**Additional information**
I use a purge bucket with start and end gcode for both.
Here is a print screen   showing the unexpected moves toward the prime tower which is unchecked.
[image]

clelong, 2019-11-07:
Loaded a Gcode that had been generated by my Cura 4.3.0.
I see that the move to the prime tower occurs on each color changing...
[image]

rburema, 2019-11-07:
> I see that the move to the [base position] occurs on each color changing...

That's correct: In our model, we assume (at least) 2 extruders/tools are needed for a color change. A lot of things happen on a tool-change. Most notably, one extruder needs to be put on a stand-by temperature, and the other one needs to be heated up from the stand-by temperature to printing temperature. You don't really want either nozzle to be _on_ the model while heating up (or cooling down for that matter).

clelong, 2019-11-07:
The problem is that there must be  an automatic small prime amount, and it creates a small oozing that spoils the print! 
Is there a way to get rid of this small oozing?

Ghostkeeper, 2019-11-07:
The move to a base position on extruder switching can be avoided by setting the `machine_extruder_start_pos_abs` and `machine_extruder_end_pos_abs` settings to `false` in the definition files, and then the `x` and `y` positions of the start and end location to 0.

You could try the [Enable Ooze Shield](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/ooze_shield_enabled.md) setting to get rid of the oozing. Alternatively, print a prime tower.

closed 2019-11-08 by Ghostkeeper