Ultimaker/Cura #6633: Hotend and bed cooling down before print finishes

lolligeaap, 2019-11-06:
**Application version**
4.3.0

**Platform**
Windows 10

**Printer**
Anycubic I3 Mega S with Marlin 1.1.9 firmware with BLTouch and a E3D V6 hotend. 

**Reproduction steps**
The hotend starts to cool down right before the end of the print (97/98%). The printer continues to print but the filament can't go through the nozzle when it cools down. This happens in the same place with the same model. 

**Actual results**
The printer stops heating the hotend and the hotbed and the print fails at the end. In the Octoprint terminal you can see the M140 S0*84 and M104*86 codes being sent to the printer while still printing the model. I can't find these codes in the original Gcode file.
[image]

**Additional information**
I checked the connections of the hotend and the thermistor, but they seem fine. The thermistor still works, no strange behavior.

rburema, 2019-11-07:
>  I can't find these codes in the original Gcode file.

What do you mean by 'the original gcode file'? The file we produce when 'save to <...>'? That's the same as the data we send to the printer. (Otherwise a file printed from an USB disk would produce different results than over the network.)

lolligeaap, 2019-11-07:
> > I can't find these codes in the original Gcode file.
> 
> What do you mean by 'the original gcode file'? 

The file that Cura creates when slicing the model. Printing from SD or directly from Octoprint gives the same result; failed print at the end.

Ghostkeeper, 2019-11-07:
The 3D printer typically keeps a buffer of motion commands. It pre-parses those to ready them for execution. I know that Marlin used to execute heating commands out of band at some point so it would do heating and cooling too early. Maybe this has been fixed in recent Marlin versions. If that is your problem you could try putting a pause command of a few seconds in the end g-code.

But since the original g-code file that Cura produces is correct, I don't think that Cura can do anything about your problem. The problem is in your printer's firmware.

closed 2019-11-07 by Ghostkeeper

dhysuiej, 2023-06-10:
@lolligeaap I'm having a similar issue with Jyers (Marlin firmware fork) on the Ender 3 v2. Did you solve the issue upgrading to a newer firmware version?

lolligeaap, 2023-06-10:
> @lolligeaap I'm having a similar issue with Jyers (Marlin firmware fork) on the Ender 3 v2. Did you solve the issue upgrading to a newer firmware version?

I’m sorry but I can’t remember, it’s been a while ago. I don’t think I upgraded the firmware, maybe it had something to do with the slicer or the way the slicer generated G-code.