Ultimaker/Cura #6634: No postprocessing happening on OSX.

hedefalk, 2019-11-06:
**Application version**
4.3.0 
**Platform**
OSX

**Printer**
UM2

**Reproduction steps**
1. Adding a post-process to pause at height
2. Slice

**Actual results**
No postprocessing.

The vactrol1 file is without the PauseAtHeight and vactrol2 is with the plugin enabled

```
✗ diff ~/Desktop/temp/UM2_vactrol1.gcode ~/Desktop/temp/UM2_vactrol2.gcode
11a12
> ;POSTPROCESSED
```

**Expected results**
I expect actual GCode to be inserted.

** Additional info
See nothing but this in log:

```
tail -1000f  ~/Library/Application\ Support/cura/4.3/cura.log | grep ERROR

2019-11-06 22:43:59,700 - ERROR - [MainThread] LocalContainerProvider.LocalContainerProvider.saveContainer [91]: Failed to get MIME type for container type [<class 'cura.Settings.PerObjectContainerStack.PerObjectContainerStack'>]
```

hedefalk, 2019-11-07:
Ooh, it was in fact that all the plugins I tried with setting layer height in mm doesn't work. It works for an integer layer height, but not for defining the height in mm. They all just silently fail in this case. Updated header

Ghostkeeper, 2019-11-07:
Looks like it just pauses too late? I tried to slice a cube with the pause height set to 5mm (the default). It produced this g-code:

```
G0 X160.56 Y124.44
G0 X160.8 Y124.2
;TIME_ELAPSED:624.086894
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current z: 9.1
;current height: 9.1
M0 ; Do the actual pause
;LAYER:89
;MESH:cube.stl
G0 X160.8 Y124.2 Z9.1
M204 S1000
M205 X10 Y10
;TYPE:WALL-INNER
G1 F1800 X160.8 Y115.8 E97.43324
G1 X169.2 Y115.8 E97.47274
```

hedefalk, 2019-11-07:
I got nothing, see my diff. I paused a 7mm print at 3.5. Overshoot? But why?

Ghostkeeper, 2019-11-07:
If the pause happens above the print, that could happen. Like overshoot, yeah.

But that it pauses at the wrong height is definitely buggy.

b-morgan, 2019-11-19:
I believe you will discover the problem if you modify the PauseAtHeight.py script to add a couple of lines, one to print the layer_0_z (which I suspect is the problem) and one to print the pause_height (for a sanity check):

`                prepend_gcode = ";TYPE:CUSTOM\n"
                prepend_gcode += ";added code by post processing\n"
                prepend_gcode += ";script: PauseAtHeight.py\n"
                if pause_at == "height":
                    prepend_gcode += ";layer_0_z: {z}\n".format(z = layer_0_z)
                    prepend_gcode += ";current z: {z}\n".format(z = current_z)
                    prepend_gcode += ";pause height: {height}\n".format(height = pause_height)
                    prepend_gcode += ";current height: {height}\n".format(height = current_height)
                else:
                    prepend_gcode += ";current layer: {layer}\n".format(layer = current_layer)
`

nallath, 2019-11-22:
This has been fixed by https://github.com/Ultimaker/Cura/pull/6690

closed 2019-11-22 by nallath