Ultimaker/Cura #6638: Auto scale parameter on model import

Jollar22, 2019-11-07:
No auto scaling when importing models. To correct material shrinkage, I use 2 parameters: MODEL SCALING and HORIZONTAL EXPANSION. Each time for different materials I must remember to enter these parameters manually.

It would be convenient if you could add auto scaling in the material profile on model import and the horizontal expansion parameter. Then it would be enough just to select the material, print profile, import models and press "Slice". And get superior print accuracy.
At worst, you can make a global auto-scale option when importing a model.

I think that this feature will make Cura to the top from slicers.

rburema, 2019-11-08:
Hi @Jollar22,

Concerning the 'horizontal expansion' part of your request; there is already a way to do this, sort of. You can change the 'horizontal expansion', then 'create a profile from current settings/overrides...'. This will of course be a profile that isn't linked to any specific material, so you'll have to create a new profile for each material.

Concerning the rest of the proposal; I'll discuss it with the team.

cheers, Remco.

Jollar22, 2019-11-08:
Thanks, I do so. Auto scaling is so lacking!

rburema, 2019-11-08:
It seems it will be some time before we can decide if, when and how to take this up, as it is closely related to requests some other teams had. We'll try to update this issue when we have more information.

maukcc, 2020-04-04:
If we could just get an option to do horizontal expansion in %, that would be the solution.
As shrink is in % and not in mm.
That would help a lot.

nallath, 2020-04-06:
We're looking into getting a shrink factor that will scale the model in all directions, not just the horizontal expansion.

maukcc, 2020-04-06:
please make Z scale factor a different setting, as it is not related to X and Y

Ghostkeeper, 2020-04-07:
We know that it shrinks differently in the Z direction. Details aren't fleshed out yet, but I think we'll find it necessary to have a different shrinkage ratio in the direction of the layer stacking soon enough. Nothing quite like proper testing though.

ddosoff, 2021-06-02:
Scaling for x, y and z is good option for material