Ultimaker/Cura #6639: Feature Request: fan "kick" for fan speeds below 100%

javascriptsnack, 2019-11-08:
When building up to the maximum fan speed, the cooling fan won't start immediately on the Ender 3 (and maybe other printers with similar squirrel fans). I suspect this is because, at lower power, the fan can't overcome the static friction of its bearing. For example, if the fan speed is at 25% then fan won't spin, however, if it's flicked in the right direction it will start spinning and continue spinning. 

The simple solution I envisioned would be a "kick start" checkbox under the cooling settings or maybe in the experimental settings on Cura. If checked, the fan would turn on 100% for ~0.5 seconds the first time it was turned on, and then continue at whatever speed it should be at. For example, if layer 1 is 0%, layer 2 is 50% and layer 3 is 100%, the fan wouldn't turn on at layer 1, would turn on at 100% for 0.5s at layer 2 and then continue at 50%, and would turn on 100% for layer 3. 

Alternatives might be a custom plugin or having the feature be implemented Cura by default. Most likely, the kick can be a lot shorter than 0.5 seconds long so it shouldn't accidentally create too much airspeed and cooling. The kick start time could also be adjustable by the user to account for different fans with different requirements.

nallath, 2019-11-08:
Ideally, this should be something that the firmware handles. In quite a few situations, this is already the case.

rburema, 2019-11-08:
We've decided not to work on this. As @nallath says, this should be handled by the printer-firmware.

While on the one hand I'm closing this ticket, we're not, on the other hand, so against it that we won't consider pull requests that address this issue.

closed 2019-11-08 by rburema