Ultimaker/Cura #6641: Is there a setting for this?

FlailAway, 2019-11-09:
**Application version**
4.4.0 BETA

**Platform**
Mint 19.2+Cinnamon

**Printer**
Any

**Reproduction steps**
See pics

**Actual results**
Experimental Adaptive Layers: The thinner layers quit too soon.

**Expected results**
The thinner layers should extend farther at the top and bottom of the holes. Possibly a little past the top and bottom layers of the holes?

[image]
[image]

ninovanhooff, 2019-11-11:
1: To clarify: Is it a regression, meaning this was working correctly in 4.3.0 and is now broken in 4.4.0 beta?
2: What would be the benefit in your view of having more thin layers past the top and bottom of the hole? Strength? Shape Accuracy?
3: Could you attach a 3mf project file?

FlailAway, 2019-11-11:
No idea on the Regression as I do not keep earlier appimages.

But, it is not extending far enough to be past the hole boundary, look at the lower pic and the bottom part of the hole. The tighter layers are quitting before it gets to the bottom.  The top is almost as bad as they quit before going past the top few layers of the hole.

Yes, it is a visual thing, well, kinda as it makes for some extra clean up for clean fluid flow. Also when I printed some holders for round linear bearings with a similar orientation, I had to run a drill through to make it round at the top and bottom of the holes.

The PC for the printers is not here and not online so unless a 3mf is critical, I'd like to be excused. :)

Ghostkeeper, 2019-12-03:
There is a setting that influences this yes: Adaptive Layers Variation Step Size. You can see a write-up of this setting with some screenshots in my Settings Guide: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/adaptive_layer_height_variation_step.md

When calculating the layer thickness Cura progresses from bottom to top. Then because the variation between adjacent layers is limited by Adaptive Layers Variation Step Size, it'll slowly graduate to the desired layer thickness. This makes it different at the bottom than at the top. If you increase the step size, it'll more closely match the desired layer thickness. But if you increase it too much, you'll get banding.

What the title of this issue is asking is not currently possible though. Cura will not currently "look ahead" to higher layers to see whether it needs to start reducing layer thickness. That is a bit difficult to do since to know that it'll need to create cross sections of the models but then based on those it needs to adjust the layer thicknesses, meaning those cross sections are in the wrong place. We'd need to rewrite the algorithm to have a sort of global relaxation to optimise that better, which could be a lot of work.

LipuFei, 2019-12-04:
For developers, the JIRA ticket is CURA-7026