Ultimaker/Cura #6642: Multiple Process, Sequential Printing Mode

landon-hillyard, 2019-11-09:
**Application version**
4.3.0

**Platform**
Windows 10. Intel I7. 

**Printer**
Ender 3

**Reproduction steps**
N/A

**Actual results**
N/A

**Expected results**
N/A

**Additional information**
https://www.simplify3d.com/support/articles/multi-part-printing/
Simplify 3d allows me to define how many layer to print on a given model before switching to the next model. i.e. print 10 layers, z hop just more than 10 layers, then move to the next model. 

It would significantly reduce my print time if I could do this in Cura. 

Also, an experimental feature to know the print off the bed when it is done would be pretty cool.

nallath, 2019-11-11:
This was already asked for previously in https://github.com/Ultimaker/Cura/issues/1431. Some work was done on it in the past, but if I remember correctly, it quickly ran into quite a few issues / complications.

Reality4DEvolution, 2021-01-25:
i also add my name to this request.
currently, PRINT SEQUENTIALLY is allowed for items below the gantry height.  example, 10 mm.
so, say 4 objects, printed sequentially -  object 1, 0-10 mm, object 2, 0-10 mm, etc.  SO, once the 10 mm is done, suppose the objects are 100 mmm.
why not enable printing in 10 mm unites, so that next print would be object 1, 10.01-20 mm, object 2, 10.01-20 mm, etc?  and it would print everything, then start again at that last-printed-height.

in my thinking, this would be no diferent than being at bed level ZERO, going to 10 mm;  the base print level would just get reset in each set of object-sequential-printing

anyway, thanks for listening

closed 2023-07-13 by github-actions[bot]