Ultimaker/Cura #6643: Crash opening file Mac os Catalina [4.3.0] And [ 4.4]

sethiris, 2019-11-10:
**Application version**
[4.3.0] And [ 4.4]

**Platform**
MAC OS CATALINA 10.15.1
  Nombre del modelo:	MacBook Pro
  Identificador del modelo:	MacBookPro15,1
  Nombre del procesador:	6-Core Intel Core i7
  Velocidad del procesador:	2,2 GHz
  Cantidad de procesadores:	1
  Cantidad total de núcleos:	6
  Caché de nivel 2 (por núcleo):	256 KB
  Caché de nivel 3:	9 MB
  Tecnología Hyper-Threading:	Activado
  Memoria:	16 GB
  Versión de la ROM de arranque:	1037.40.124.0.0 (iBridge: 17.16.11081.0.0,0)
  Número de serie (sistema):	C02WX326JG5H
  UUID de hardware:	F7022991-7EDB-58CB-B3C6-CA93E0C6089B
  Estado del bloqueo de activación:	Activado

**Reproduction steps**
Selectin, File, Open. When the finder windows appears, it just crash

**Actual results**

**Expected results**
(What should happen after the above steps have been followed.)

**Additional information**

Process:               cura [1925]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.4 (4.4.0-BETA)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [1925]
User ID:               501

Date/Time:             2019-11-09 21:25:36.189 -0300
OS Version:            Mac OS X 10.15.1 (19B88)
Report Version:        12
Bridge OS Version:     4.1 (17P1081)
Anonymous UUID:        64400309-DD97-DC4D-5990-97D7519C33C7

Sleep/Wake UUID:       B05DBC57-F0B9-4DFC-93D1-8E4C99413F4B

Time Awake Since Boot: 1700 seconds
Time Since Wake:       1000 seconds

System Integrity Protection: disabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0:
--> 
    __TEXT                 0000000102763000-000000010290e000 [ 1708K] r-x/rwx SM=COW  /Applications/Ultimaker Cura.app/Contents/MacOS/cura

Thread 0 Crashed:
0   libsystem_kernel.dylib        	0x00007fff677b549a __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff676cd3a2 raise + 26
2   libsystem_platform.dylib      	0x00007fff67867b1d _sigtramp + 29
3   ???                           	000000000000000000 0 + 0
4   com.apple.FinderKit           	0x00007fff4ba1837d TFENode::FinderTags(bool) const + 71
5   com.apple.FinderKit           	0x00007fff4bb8158e +[FI_TListViewController updateTagsForCellView:node:] + 42
6   com.apple.FinderKit           	0x00007fff4bb7934e -[FI_TListViewController dataSourceItemsDidChange:] + 2283
[... 1520 lines cut ...]
shared memory                      660K       19 
===========                     =======  ======= 
TOTAL                             75.5G     3399 
TOTAL, minus reserved VM space     3.3G     3399 

Model: MacBookPro15,1, BootROM 1037.40.124.0.0 (iBridge: 17.16.11081.0.0,0), 6 processors, 6-Core Intel Core i7, 2,2 GHz, 16 GB, SMC 
Graphics: kHW_IntelUHDGraphics630Item, Intel UHD Graphics 630, spdisplays_builtin
Graphics: kHW_AMDRadeonPro555XItem, Radeon Pro 555X, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E1
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E1
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Oct 19 2019 15:33:10 version 9.113.2.0.32.5.39 FWID 01-5eb3e1fa
Bluetooth: Version 7.0.1f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: USB 10/100/1000 LAN, Ethernet, en7
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.3
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.3

smartavionics, 2019-11-10:
Hello @sethiris , I don't know why Cura crashed on your Mac but you may wish to try one of my Cura releases as they are built to be compatible with more recent MacOS releases. You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README file in there.

sethiris, 2019-11-10:
Thanks you for the response, but sadly it fail in the same way u__u

> Hello @sethiris , I don't know why Cura crashed on your Mac but you may wish to try one of my Cura releases as they are built to be compatible with more recent MacOS releases. You can find my builds [...]

smartavionics, 2019-11-10:
Did you try today's build (Cura-mb-master-macos-20191110.dmg) ? That uses a more recent version of Qt.

LipuFei, 2019-12-04:
For developers, the JIRA ticket is CURA-7025

Gipielle, 2020-02-10:
Hello, 
same problem with macOS Catalina 10.15.3

Gipielle, 2020-02-10:
> Did you try today's build (Cura-mb-master-macos-20191110.dmg) ? That uses a more recent version of Qt.

Hello, sorry how do I build the dmg ? Thanks

smartavionics, 2020-02-10:
That particular build has now  been deleted. Try using the latest release (Cura-mb-master-macos-20200208.dmg).

Gipielle, 2020-02-10:
Thanks, where i can find Cura-mb-master-macos-20200208.dmg ?

smartavionics, 2020-02-10:
See my replies above...

Gipielle, 2020-02-10:
Found, this works ! Thanks

wwwiniew, 2020-03-23:
I found that if I cd via terminal to the .app directory and execute cura via sudo it opens the file menus without crashing.  I have the privacy set to allow full disk access yet this crashes without sudo.

closed 2024-12-07 by github-actions[bot]