Ultimaker/Cura #6645: [4.3.0] Ender 3 nozzle moving down into primed line

baddogg1231, 2019-11-10:
Hey Cura team. Love the program and have been using it as the daily driver for all my printing needs since the beginning. With the 4.3.0 update, I've run into a [similar issue as another user that was posted here](https://github.com/Ultimaker/Cura/issues/6424), but this is for the Ender 3 and I was not happy with the workaround provided in that thread.

I use the popular Ender 3 prime line that is added in the 'Start G-Code' section of Cura.

Specifically: 

; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

This has worked beautifully for me with all versions of Cura up until 4.3.0.

After reading the thread I attached, I see that the print head moves to Z0 before moving to the start of the actual print, but that causes my prime line (and others who use this popular start code) to no longer function as the print head moves down into the line, which makes it stick the print head, and is dragged along the print bed and gets in the way of the print.

The solution to move the print head over a bit then down before printing is an okay solution but I think the way it was done before to be much better. However, I have modified the start g-code to include a similar function to the edit of the i3-mega so that it will be updated with the next version of Cura in a [pull-request here](https://github.com/Ultimaker/Cura/pull/6644)

rburema, 2019-11-12:
Hi @baddogg1231 

We've noticed several problems with the new behaviour introduced in 4.3 as well, and there is at least one internal ticket open to address these problems. We're very close to the release of 4.4 though, so it's uncertain if that can be in there.

> [...] so that it will be updated with the next version of Cura

That goes for your PR as well I'm afraid; the beta for 4.4 is already out, so that'll probably be 4.5 (if we do decide to pull in in that is).

(Please also see @nallath's commentary on the PR.)

baddogg1231, 2019-11-12:
> Hi @baddogg1231
>
> We've noticed several problems with the new behaviour introduced in 4.3 as well, and there is at least one internal ticket open to address these problems. We're very close to the [...]

Definitely glad to see the issues are being worked on internally! My PR is more of a workaround rather than an end-all solution. The original start g-code in 4.2 versions worked great as it kept the print head away from the bed during the initial travel move (which I understand shouldn't be a problem, but I like the fail-safe in case of too fast accelerations), and I understand that changing to move to Z-home before starting the print could be beneficial in most cases, but with the start g-code prime I've always used, it seemed to work better for me for priming and cleaning the nozzle before starting the print, and prevented the plastic from being drug across the print surface into the print.

If the change is reverted or somehow mitigated in any of the upcoming releases, then the g-code commit I made, minus the "Move over to prevent blob squish" line, still works extremely well in my case.

If by chance the PR does get added in some form, I believe it could be beneficial to many. I replied to @nallath's comment on my PR for possible further testing on Creality machines, but it is confirmed to work on my Ender 3.

Ghostkeeper, 2019-12-03:
I can't reproduce this in Cura 4.4. This is a fragment of the g-code near the beginning, with my added notes:

```
G1 Z2.0 F3000 ;Move Z Axis up     <--- Last line of start g-code, leaves nozzle at Z=2.0

G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:50
;LAYER:0
M107
G0 F6000 X97.198 Y97.264 Z0.2     <---- Diagonal move towards the centre and Z=0.2
;TYPE:SKIRT
G1 F1500 E0
G1 F1200 X97.886 Y96.627 E0.03119
```

So it looks like the move is diagonal to the centre now, up to the initial layer height. That should work fine.

Could you try again with 4.4? We've done some investigation on this part so maybe it was fixed in the meanwhile.

nallath, 2020-02-20:
Closing this due to inactivity.

closed 2020-02-20 by nallath

Muskateer10, 2020-03-03:
It would be nice if there were 2 options for the prime line. 
1) It would be really nice if it showed up in preview mode, so we are made aware if the print were to print on top of it.  I have a print that is 234mm wide, I did not catch the fact that the prime-line was underneath it. My bed is 235mm wide. I prefer to print in the X direction for wide prints. 
2) It would be nice to be able to easily move the prime line location, either to the back (xmin+10mm to xmax -10mm) or something similar.