Ultimaker/Cura #6646: Change at Z printer speed specified value.

photodude, 2019-11-10:
**Is your feature request related to a problem? Please describe.**
In the post-processing change at Z 5.1.1 settings you can set the speed and printer speed as a percentage. It would be very helpful for speed tuning tests to be able to specify a specific value and not a percentage. Think temperature tower but with print speeds as specified values.

see image showing the current percent values
[image]

**Describe the solution you'd like**
give an option for specified speed settings 

End result should allow very clear progression of exactly specified values something like this for the set of postprocessing gcode from change at z. Like a temperature tower but with print speeds as the specified values i.e. a speed tower.
initial print speed 40mm/s (0mm on model)
- increase by 5mm
  - new print speed 50mm/s
- increase by 5mm (now at 10mm on model)
  - new print speed 60mm/s
- increase by 5mm (now at 15mm on model)
  - new print speed 70mm/s
- increase by 5mm (now at 20mm on model)
  - new print speed 80mm/s
- increase by 5mm (now at 25mm on model)
  - new print speed 90mm/s
- increase by 5mm (now at 30mm on model)
  - new print speed 100mm/s

**Describe alternatives you've considered**
1.  replace existing percent speed with exact values
2.  show both options and calculate the other option
  - i.e. calculate speed from entered percentage, or if speed is entered calculate the percentage

**Affected users and/or printers**
users who like to tune their slicer settings to get the most out of their printer
manufacturers developing printer profiles for their printers

**Additional context**
a random Thingiverse model for this speed tower concept
https://www.thingiverse.com/thing:3249901
we just lack a very precise way of specifying the speeds for the changes.

rburema, 2019-11-13:
Hi @photodude ,
Sadly, we've chosen to defer this for now, though pull requests are of course welcome.
cheers, Remco.

photodude, 2019-11-13:
@rburema point me in the right direction and I'll make an attempt to implement the change. I need to know where to look, what to look for, and what Unit test changes would need to be made/implemented. I consider my python skills very introductory as I've only used it for numerical methods with FEA in jupyter notebook.

rburema, 2019-11-13:
The relevant file is: https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/ChangeAtZ.py

P.S. It's a script/plugin, so maybe you won't have to build Cura from source (not really an easy process yet...)

P.P.S. Python itself is not that difficult to learn. Qt / QML might be a little bit trickier, but you would just have to copy-and-alter existing code basically.

photodude, 2019-11-17:
@rburema I don't disagree that python is somewhat easy to learn, I understand basic python and basic programming (c++, matlab, R, arduino, EES, PHP), but parsing what that contextually means in relation to this program/script to make a change is different.

There is a level of parsing and comprehending the API classes and variables used in the script/plugin and what all the intermixed gcode means.

Below is my attempt to understand the code...

Looking through the script there are 55 references to `speed`; of those 55 instances only some of those relate to the print speed, others relate to fan speed. Apparently I have to search older versions of cura for this script as the full history is not connected to this file; so currently I can't take contextual clues from when the print speed adjustment was added without hunting down the relevant older file version (assuming there is a clear history of when speed changes were added).

I can tell that  M220 has something to do with the relevant speed change

[Lines 112-149](https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/ChangeAtZ.py#L112-L149) are the relevant input fields 123 and 142 are the lines with unit `%` which need to be changed

[lines 305-334](https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/ChangeAtZ.py#L305-L334) seem to be some type of state change check

lines [315-323](https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/ChangeAtZ.py#L315-L323) 423 and [435+436](https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/ChangeAtZ.py#L435-L436) seem to be relevant calculations (in something like sprint format) I assume only the part `(int(f / 100.0 * float(target_values["printspeed"]))` is relavant.

but for the most part I'm not yet seeing how this particular function is working. all I see is the large pattern for what it does..... check if user wants to change something, based on what and where to make a change -> modify the gcode.  reading this code is not yet clear for me in a way to look at the code parse it's meaning and make the necessary modifications.

I think I could easily follow the design pattern in the script to add a new field(s) that would allow user input for specified changes to print speed. but I'm not sure how to set up the logic control structures with all the if statements (there are a lot of checks and conditions). After that it would just be replacing/inserting the speed value in the `modified_gcode` (and somehow wrap it in a control structure to avoid stepping on the toes of the old calcs, but that would likely be confusing to users.

Any guidance on lines to look at for these changes would be helpful.

Ghostkeeper, 2019-12-03:
One major factor in deferring this issue is that the code in that script is pretty complex to understand, not easy to modify. We've inherited the script and have been keeping it passably up-to-date but nobody in the Cura team has ever made major changes to it, so our understanding of it is limited as well.

> Apparently I have to search older versions of cura for this script as the full history is not connected to this file; so currently I can't take contextual clues from when the print speed adjustment [...]

This history goes back into Legacy Cura: https://github.com/daid/LegacyCura

I think the most major change you need to make is at line 435+436. There you'll need to output different values for F. Which value you're going to output needs to depend on the settings.

I think if we are to accept your changes in, it needs to still allow the current workflow as well, so you'd need to have boolean or enum type settings to select how the speeds are adjusted (by ratio, to a fixed value or with a delta).

closed 2023-07-13 by github-actions[bot]