Ultimaker/Cura #6648: [bug] hard frieze every changes

userosos, 2019-11-11:
Cura 4.3. If i changes an option (any) i has had hard frieze. For example i remove an figure in retract after it i has not respond from cura about 60 sec after i set the setting to 6 and has not respond again.
My laptop: Nitro AN515-52, i5-8300, gtx1050 (cura use it).
Log:
https://dropmefiles.com/YzzFM

nallath, 2019-11-12:
Have you also tried the 4.4 beta? I've done some performance improvements there.

rburema, 2019-11-12:
Hi @userosos 

I see that the RepetierIntegration plugin is spamming the logs a lot (with what appears to be requests to  an internal IP), could you try disabling that to see it that's the cause?

userosos, 2019-11-12:
@rburema 
> 
> 
> Hi @userosos
> 
> I see that the RepetierIntegration plugin is spamming the logs a lot (with what appears to be requests to an internal IP), could you try disabling that to see it that's the cause?

I disabled RepetierIntegration  (remove the folder from plugins folder). And it work OK.

@nallath ok. I will tast it but RepetierIntegration plugin nit work in cura 4.4 (i think).

userosos, 2019-11-12:
I test Cura 4.4 now but the RepetierIntegration not work now. I changed the plugin.json file to:
{
    "name": "Repetier Integration",
    "author": "VMaxx - c/o fieldOfView",
    "version": "4.4",    
    "minimum_cura_version": "4.0",
    "maximum_cura_version": "4.2",
	"maximum_cura_version": "4.3",
	"maximum_cura_version": "4.4",
    "description": "Enables network printing and monitoring to Repetier",
    "supported_sdk_versions": ["6.0.0","6.1.0","6.5.0","7.0.0","6.3.0","6.4.0"],
    "i18n-catalog": "cura"
}
It not helped me.

rburema, 2019-11-12:
Perhaps the ones that make the RepetierIntegration plugin can help you out: Their issue tracker is over [here](https://github.com/VMaxx/RepetierIntegration/issues).

They probably haven't updated for 4.4 yet, since it's quite new and (I could be mistaken but I think) they're not in the official program (so they don't get the contributor emails to notify them).

Ghostkeeper, 2019-12-03:
We're going to assume that the problem is in the RepetierIntegration plug-in then.

closed 2019-12-03 by Ghostkeeper