Ultimaker/Cura #6649: 4.3.0 Middle Raft Layer Line Width / Spacing Not Updating

fdmmaniac, 2019-11-11:
I am trying to decrease spacing between lines on the Middle Raft Layer, but the print does not change.  The Middle Raft Layer is so weak that it separates from the Top Raft Layers when I remove the Raft from the part.
Is there a glitch with the settings for these two Middle Raft parameters?
Cura Middle Raft Layer Settings: 0.55mm Line Width, 0.60mm Spacing (nearly solid)
Visual image in Cura 4.3.0:
[image]
Actual Print:
[image]
Basic Prusa i3 Printer
Thanks!

nallath, 2019-11-12:
What the preview shows is what toolpaths are being generated. I don't have a Prusa I3, so I can't print a raft myself.

rburema, 2019-11-12:
You can reload the produced gcode in cura as well, to double-check.

fdmmaniac, 2019-11-12:
Great.
I will try that.
Thanks,
Erik

Reply-to: reply@reply.github.com
Cc: eorwoll@nusignal.com; author@noreply.github.com

Ghostkeeper, 2019-12-03:
It seems that the problem here is not in Cura. As shown in your screenshot, Cura's g-code is correct. It's just the actual print result that is wrong. I also couldn't reproduce the issue on this side.

I think in order to achieve your desired result, you'll want to reduce both the line width and the line spacing to 0.4mm or in any case something very close to your nozzle size.

closed 2019-12-03 by Ghostkeeper

fdmmaniac, 2019-12-03:
Thank you for the help.
I would say that the screenshot is not a shot of the gcode.  It is a shot of the preview image created in Cura.
And yes, Cura shows the image correctly.
Are you suggesting the problem is with the Printer itself?  (Firmware, etc?)
-Erik

Ghostkeeper, 2019-12-06:
> I would say that the screenshot is not a shot of the gcode.  It is a shot of the preview image created in Cura.

Yeah, true. Sometimes that preview is wrong (see also #6663). But mostly it's reliable enough, especially when directly slicing in Cura (as opposed to reading out a .gcode file)

I think the problem is probably not in the firmware, but in the feeder or the Z stage, combined with the settings being used in Cura. Try a smaller line width because it seems like the feeder is not keeping up with the amount of extrusion requested.