Ultimaker/Cura #6650: Cura 4 (4.3 & 4.4B) Add Printer Menu Scrolling Issue

roseg0ld, 2019-11-11:
**Application version**
I've noticed this issue since the upgrade to Cura 4

**Platform**
This issue only appears on OSX (Macbook Pro)

**Reproduction steps**
1. Whenever I try to scroll the Add Printer Menu with my USB mouse, the scroll bar resets to the top on every roll of the scroll wheel
2. This can be avoided by clicking and dragging the scroll bar or using the trackpad's scroll functionality.

**Additional information**
I am using the Logi MX Master 2S mouse and haven't tested it with any other peripheral mice. This is kind of a strange/niche issue since as far as I can tell, it only appears on Mac with this certain wireless mouse, but it can be challenging to add a new printer with this bug.

nallath, 2019-11-12:
Well, unfortunately we don't actually have that mouse, so it's going to be a bit tricky for us to reproduce it.

roseg0ld, 2019-11-12:
Update to the issue, it also appears scrolling the marketplace. I'm happy to do any testing and report back if you'd like!

fieldOfView, 2019-11-12:
It would help if you could see if it also happens with a different mouse.

smartavionics, 2019-11-12:
FYI, I just tried using a Microsoft wireless mouse on a MacBook Pro running High Sierra with one of my Cura releases and the mouse scroll wheel worked just fine in both the printer list and also the marketplace.

rburema, 2019-11-13:
This is a bit of a long shot, but there may be something in the logs?

Yian-Lai, 2019-11-13:
I tried using Logitech M100 and B100 mouse on MacOS Mojave and could not reproduce it on the add printer list nor Marketplace.

closed 2020-06-19 by no-response[bot]

khaoskosmos, 2020-11-25:
I also have this issue with the Logitech MX Master 2S.
[CuraScrollingBug.mp4.zip](https://github.com/Ultimaker/Cura/files/5598025/CuraScrollingBug.mp4.zip)

Nothing was added to the log while scrolling.

Ghostkeeper, 2020-11-27:
We're looking for general QML errors here really. I wouldn't necessarily expect something to happen during scrolling.

techdragon, 2021-01-25:
@Ghostkeeper This behaviour happens to me when scrolling with my 100% stock Apple Magic Mouse.

khaoskosmos, 2021-02-15:
I have yet to encounter any other app that has this glitched scrolling behavior, odd.

PoiMechanic, 2021-03-07:
Happens to me as well and it drives me nuts. Only happens in Cura program 4.7 and 4.8. Especially frustrating when you have a ton of profiles to go thru. MX Masters 3 Mouse on brand new Macbook Pro with Catalina

Ghostkeeper, 2021-03-15:
[image]