Ultimaker/Cura #6651: Cura 4.3.0 -G-Coda commands inserted *before* startup G-Code

chjohans, 2019-11-11:
This behavior was observed with Cura 4.3.0 for Windows. My printer is a Anycubic Mega-S.

I was using a material_bed_temperature of 70 degrees C and a material_bed_temperature_layer_0 of 75 degrees C.

My Startup G-code contains code to heat up both my nozzle (M104 S{material_print_temperature}) and my bed (M140 S{default_material_bed_temperature}) so my auto levelling procedure can be done with a heated bed.

The problem is Cura will insert some of it's own G-Code before my startup G-code, in this case it inserted G-Code to heat up the bed to {material_bed_temperature_layer_0} as seen in the code snippet below. The lines "M140 S80", "M105" and "M190 S80" are inserted by Cura:

`;FLAVOR:Marlin
;TIME:16737
;Filament used: 13.0963m
;Layer height: 0.2
;MINX:51.208
;MINY:90
;MINZ:0.2
;MAXX:158.792
;MAXY:120
;MAXZ:70
;Generated with Cura_SteamEngine 4.3.0
M140 S80
M105
M190 S80
M82 ;absolute extrusion mode
M501 ; Restore settings from EEPROM.
G21 ; Metric values.
G90 ; Absolute positioning.
M82 ; Set extruder to absolute mode.
M107 ; Start with the fan off.
M104 S250   ; Set hotend temperature.
M140 S75  ; Set bed temperature.
; Uncomment the next 7 lines if you are using BLTouch/3DTouch
M117 Levelling bed… ; Set status on LCD screen.
M280 P0 S160 ; BLTouch alarm release.
G4 P100 ; Delay for BLTouch.
G28 ; Auto home.
M190 S75  ; Set and wait for bed temperature.
G29 ; Enable Auto Bed Levelling (ABL).
M420 S1 ; Enable BLTouch.
M117 Cleaning nozzle… ; Set status on LCD screen.
G28 X0 Y0 Z0 ; Move X/Y to min endstops.
G0 Z0.15 ; Drop to bed.
G92 E0 ; Zero the extruded length.
M109 S250   ; Set and wait for hotend temperature.
G1 X40 E25 F200 ; Extrude 25mm of filament in a 4cm line.
G92 E0 ; Zero the extruded length again.
G1 E-1 F500 ; Retract a little.
G1 X80 F4000 ; Quickly wipe away from the filament line.
G5 ; Enable resume from power failure. Only for print from SD card!
M117 Printing… ; Set status on LCD screen.
G92 E0
G92 E0
G1 F2400 E-4.5
;LAYER_COUNT:350
;LAYER:0`

Cura correctly inserts "M140 {material_bed_temperature}" just before the second layer, the problem here is that the G-code to heat up the bed to {material_bed_temperature_layer_0} is inserted before my Startup G-code. In fact I can't think of anything (except commens) that Cura should insert before my Startup G-code.

Since I'm using {material_bed_temperature} when automatically levelling my bed the initial layer bed temperature is effectively overridden. And I can think of a whole lot of other complications, depending on what my Startup G-Code contains.

Please change this behavior so that Cura *only* inserts G-Code after the "Startup G-code".

rburema, 2019-11-12:
This seems to be duplicate. Quoting from #6633 :

> In order to thwart this safety feature, make sure your start gcode snippet contains the strings {material_print_temperature_layer_0} and {material_bed_temperature_layer_0} somewhere. It could be in a comment, or in an actual preheating sequence.

closed 2019-11-12 by rburema

fieldOfView, 2019-11-12:
That quote is actually from #6559, which by itself is a duplicate of a fair amount of reports, all the way back to #1129