Ultimaker/Cura #6652: "slice automatically" turns always ON

madias123, 2019-11-12:
**Application version**
4.3.0

**Platform**
Windows 10

**Printer**
doesn't bother

Installed Cura 4.3.0. (Updating the first time with serious problems, I totally deleted all older cura profiles/folders/items...)
Now the most annoying "feature" "slice automatically" turns automatic. There is no need in closing or reopening Cura - after a short time (or hitting "sliceing" these settings are on again.
Please fix this or remove  "slice automatically" - nobody will need this feature. 
Thank you
regards
Matthias

rburema, 2019-11-13:
Just to be sure, you have unchecked it in 'Preferences' > 'Configure Cura...' > '[ ] Slice automatically', right?

If so: Could you check what's in your cura.cfg file? The line `auto_slice = True` should not appear. You can find that file in `%appdata%/cura/4.3`.

PS:
> nobody will need this feature.

Of course it should turn off when you've disabled the setting, however: While I'm not a fan of it personally, a few of my co-developers swear by it, and it's generally well liked as far as I understand.

Ghostkeeper, 2019-12-03:
I'm unable to reproduce this problem. I've tried starting from factory defaults the following things:
* Clicking slice.
* Restarting Cura.
* Enabling then disabling autoslicing.
* Installing the Automatic Slicing Toggle plug-in from the marketplace.

After each of these steps I modified the scene (by loading a model or moving it). Then waited for half a minute. But it wouldn't automatically start a slice.

Could you post a log and the reproduce steps, like we ask in the template?

closed 2020-06-19 by no-response[bot]