Ultimaker/Cura #6656: Print Settings for Supported Overhang

vjvarada, 2019-11-14:
# Problem:
When printing with soluble supports, materials like ABS and Nylon like to curl up when being deposited on top of the soluble supports, as they default to normal printing speeds. 

# Possible Solution: 
Use initial layer settings, like larger extrusion width and/or slower print speeds may fix the problem. 

I've played around with the overhang wall and bridge settings, but it seems like the moment supports are added, the overhangs are not recognized anymore, and these settings don't apply.  

# Additional Information:
The default settings do seem to work with PVA and PLA, TPU and PVA, but Nylon and ABS seem to curl with ePVA+ from esun ( that supposedly works as support material for ABS and NYLON). Printing at overall slower speeds does solve the curling, but makes the overall print time slow, and so, have been playing with slowing down the feed rate manually on the printer when required. 

I wonder if these having these settings can even improve same material support removal.

rburema, 2019-11-14:
> I've played around with the overhang wall and bridge settings, but it seems like the moment supports are added, the overhangs are not recognized anymore, and these settings don't apply.

Based on my understanding of what you're saying:

If I change the overhang wall angle to 20 degrees, and the overhang speed to something very low (so it'll be visible) like 5%, I see no difference in the feedrate between supported and unsupported. (Set the color scheme to feedrate in the preview to check this for yourself.)

The curl up could be because the different airflow around the already build-up support may cause a local increase of heat?

vjvarada, 2019-11-14:
Yes, @rburema , If that overhang becomes supported by support material, cura stops using overhang settings.  I've turned all the fan-related settings off (fan speed override, fan speed for eEPVA+ etc), and the chamber is enclosed, with the bed going to 110C, so im sure airflow isn't an issue. Ive been able to get successful prints by lowering the print speeds at the supported skin interface.

[image]
(Above)File to test, PLA in Yellow, PVA in Purple

[image]
(Above)What the print speeds look like. Uses the normal wall and skin speeds and flow 

[image]
(Above)Same print without support. Uses the bridging, overhang wall speed settings.

vjvarada, 2019-11-14:
Update:

I turned bridging off, and just using the overhang wall speed settings am getting the following behavior when supports are off ( as expected)
[image]

When I turn supports back on, then it ignores the overhang wall settings. 
[image]

rburema, 2019-11-15:
Disregard my previous comment please, I accidentally used the wrong model to test this yesterday. Sorry!

I'll discuss it with the rest.

vjvarada, 2019-11-15:
No problem @rburema . I've made progress testing out the theory using the cura build from @smartavionics since it had the feature to enable bridge settings over support. I've posted the results at #3340 .  I've printed a few larger ABS prints since then, getting impressive results so far with no warping or curling from the ePVA+. I'm now tempted to try ABS with regular PVA to see if it prints well since it costs much lower than ePVA+. 

[image]

Ghostkeeper, 2019-12-03:
So far the Overhanging Wall experimental feature indeed doesn't consider things overhanging if they are supported, either by the model or by support. This is expected behaviour. I interpret this issue as a request to change that, i.e. to remove [this exception case in our code](https://github.com/Ultimaker/CuraEngine/blob/01061883b6b59b60f439df40af1cc51b5cb90be2/src/bridge.cpp#L52-L91) and let it bridge there.

The problem of warping above support is more of an issue with adhesion to the support. Normally the adhesion between the model and the support is weak, so that you can break off the support. In your case that's also the case but more because they are different material types that don't stick well together. When the ABS is not held down by the previous layer, it'll warp upwards.

vjvarada, 2019-12-04:
>  I interpret this issue as a request to change that, i.e. to remove this exception case in our code and let it bridge there.

Thanks for the reply @Ghostkeeper . Yes, I can see why one would need it to be as-is for printing with same-material support, but for printing with support with a different material it would improve the reliability drastically. 

Possibly have an option to enable "Overhang & bridging over support" like in the build from @smartavionics ?

Using his build I have been using bridging settings to improve reliability. I had to bump up the flowrate as well as reduce speed to get better adhesion between soluble support and model, similar to the settings for the initial layer of the print on the print bed.

LipuFei, 2019-12-04:
Hi @vjvarada , we decided not to do this now after some discussion. The request tiself is small, but there's some technical reasons behind why bridging is excluded from the supported areas. I personally am not aware of it. If you want to discuss the details, I think @Ghostkeeper and @smartavionics probably can answer your questions.

Thanks for reporting this issue. I will close it now.

vjvarada, 2019-12-06:
Thanks @LipuFei, I can definitely see why using the bridging settings for supported overhangs is a stop-gap measure, as by definition bridges won't have support under them. 

I do believe, however, that supported overhangs need to have settings of their own, independent of bridging or overhangs without support, to fine-tune reliability and performance of support.  

There is an already option for "Supported Skin Fan Speed", to which settings like "Supported Wall Speeds" "Supported Infill Speeds", and "Flow Rate for Supported Features" could be added. 

Should I create a new feature request for this?

Ghostkeeper, 2019-12-06:
No, this issue would be fine for that. It contains the correct description already. I'll re-open and propose again. But I think it's not going to get any priority. We'll see what people think.

Goodfeat, 2019-12-24:
I was sent from #6848 to this discussion.

@smartavionics  created an assembly in which there is an option "Overhanging Wall Fan Speed". 
Thank you.

[image]

Will an option "Overhanging Wall Fan Speed" be added to the official assembly?

smartavionics, 2019-12-24:
> Will an option "Overhanging Wall Fan Speed" be added to the official assembly?

Perhaps, although it is based on functionality that has already been refused for incorporation into UM's releases by the UM developers so probably it will only be in my releases. Sorry about that.

Goodfeat, 2019-12-24:
I tried to test Cura-mb-master-win64-20191223.exe.

I liked how the "Overhanging Wall Fan Speed" function works.

However, for some reason, the function starts to work if the "Overhanging Wall Speed" setting is <100% (there is no logic in this)

In general, the function is workable and is configured very flexibly. helps a lot when printing overhangs without supports (especially when turning on adaptive layers)

A big request from developers to include the "Overhanging Wall Fan Speed" feature in the main build version.

smartavionics, 2019-12-24:
> However, for some reason, the function starts to work if the "Overhanging Wall Speed" setting is <100% (there is no logic in this)

The logic is currently such that if the overhang wall speed is within +/- 20% of the normal wall speed it doesn't modify the fan speed.

Goodfeat, 2019-12-24:
That’s the problem.
it’s more logical to enable the function using the “Overhanging Wall Angle” <90

Overhanging Wall Angle = 40;
Overhanging Wall Speed = 100;
Overhanging Wall Fan Speed = 100:

In this case, the walls of the overhanging walls will be printed with 100% airflow, with a default speed
Such settings is as flexible as possible.

But in your version now it looks like a bug
If you put the following settings:

Overhanging Wall Angle = 40;
Overhanging Wall Speed = 100;
Overhanging Wall Fan Speed = 100:

unfortunately nothing will work(

But at the moment i have to lower the speed by 1% of the nominal so that the function turns on

Overhanging Wall Angle = 40
Overhanging Wall Speed = 99
Overhanging Wall Fan Speed = 100

It seems the option with the "wall angle" looks more promising.  But I can be wrong. Sorry

smartavionics, 2019-12-24:
No, I think you are probably right here. I don't use this feature myself so it's quite possible that I didn't do the right thing. I will look into making it behave more like your suggestion.

smartavionics, 2019-12-24:
I have made some changes. In my next release, the fan speed for overhangs will not be affected by the print speed. Also, I fixed a bug whereby some fan speed changes would be ignored. Next build will probably be later this week.

Goodfeat, 2019-12-24:
Thanks.
Is it normal that "Marketplace" doesn't work in your builds?

smartavionics, 2019-12-24:
No, it should work as normal, does your cura.log file show any diagnostic message that could indicate what the problem is?

Goodfeat, 2019-12-25:
Found a reason. Some lines were missing in my custom style file

[image]

P.S.
I hope that your improvements will interest the developers of the ultimaker

smartavionics, 2019-12-26:
Hi @Goodfeat , today's build now uses the overhang fan speed even when the overhang print speed is not altered. Will be available on dropbox later.

Goodfeat, 2019-12-26:
Excellent. I will definitely test it