Ultimaker/Cura #6657: 4.3 and 4.4B Bugs with Screen sizing and multiscreen setups.

DX12Diodes, 2019-11-14:
**Application version**
4.3 and 4.4B

**Platform**
Windows 10, i7 4ghz cpu, 16gb ram, nvidia 1070gtx gpu, 3x 4k Dell monitors.

**Printer**
Ender5

**Reproduction steps**
Issue1 (4.4B only)
1. Load Cura
2. Attempt to resize the window.

Issue2
1. Load Cura.
2. Move the Cura window to another screen that isn't the main screen.
3. Close Cura.

**Actual results**
Issue1
All the menus disappear from the window as per attached photo so the window isn't scaleable at all. On a 4k screen its only about 1/4 of the screen so not being able to resize it is not practical. This happens even if you click the maximise button on the window.
[image]
Issue2
After closing Cura on a different window if you attempt to open Cura again it trys to open it in the screen it was closed on but never does. The application just crashes. The only way to fix this is to then remove Cura and all settings completely from the machine and reinstall.

**Expected results**
Issue1
Cura window should be scalable to any size and have all the menus available.

Issue2
Cura should be able to be moved to any screen and it continue to function properly. The screen its loaded on should not impact the application.

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

fieldOfView, 2019-11-14:
For issue 2, there is a checkbox in the General preferences labeled "Restore window position on start". If you uncheck that, this should prevent that problem.

fieldOfView, 2019-11-14:
Are all the monitors connected to the Nvidia GPU, or do you also have an embedded Intel GPU in that i7?

DX12Diodes, 2019-11-14:
All 3 are connected to the Nvidia GPU, there is an embedded GMU on the mobo but it has nothing connected to it.

rburema, 2019-11-15:
- Are the drivers up to date?
 - Do you connect directly to the monitors or do you use a dock?

DX12Diodes, 2019-11-15:
Latest NVidia driver, direct connected to the gpu no docks or anything in between.

Ghostkeeper, 2019-12-03:
I have a multi-screen set-up rendered by an NVIDIA GPU as well at home (GTX 2070 Windforce), but have not seen these issues yet. Maybe it's specific to the 1070 drivers?

closed 2024-12-07 by github-actions[bot]