Ultimaker/Cura #6659: Gaps in corners of walls when they are horizontal

ShakieVan, 2019-11-14:
**Application version**
4.3.0

**Platform**
Win10

**Printer**
custom FFF printer
Sidewinder X1

**Reproduction steps**
1: Insert the screw
2: Place it vertical
3: Slice

**Actual results**
There are gaps in the corners, where the "next" wall sits on the infill. So the part breaks at this point at low forces.
When placing it horizontal, The walls are strongly connected.
[image]

**Expected results**
The walls should connect in the corners like they do when placing it horizontal.

**Additional information**
[Cura Gap.zip](https://github.com/Ultimaker/Cura/files/3848687/Cura.Gap.zip)

smartavionics, 2019-11-14:
It's happening because that skin region is narrower than the value of Skin Removal Width. If you set that to 0 the skin region survives...

[image]

ShakieVan, 2019-11-15:
[image]

Thank You!!!

Ghostkeeper, 2019-12-03:
This is expected behaviour. It's really part of the [trade-off](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/bottom_skin_preshrink.md) when adjusting this setting.

closed 2019-12-03 by Ghostkeeper