Ultimaker/Cura #6660: UMODE BUG Cura 4.3.0 setting: Enable Support Interface activates travels outside object.

PvD77, 2019-11-15:
**Application version**
Cura 4.3.0

**Platform**
Windows

**Printer**
Ultimaker Original Dual Extruder

**Reproduction steps**
Setting the setting: Enable Support Interface
(When this setting is not used, there is no problem)

**Actual results**
Extra long travels outside the object. (not needed)

**Expected results**
Normal result, but without the extra travels.
[image]
[image]

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

Ellecross, 2019-12-02:
UM Cura devs, see CURA-7018

PvD77, 2019-12-03:
Hello Ellecross,

CURA-7018 ? Please help me further....

I have discovered that this position outside the model is the prime tower position.
Even when the setting is not activated.
When the other nozzle is activated, the head goes to the prime tower position.
And leaving directly, without building the prime tower.

If I activate the prime tower and move it. (X, Y location) 
And then deactivate the prime tower again.
Then the head goes to the new position!

Ghostkeeper, 2019-12-03:
> CURA-7018 ? Please help me further....

Sorry for the confusion. This is a reference to an internal issue in Ultimaker's tracker. It means it was put in our planning software.

The extra information that it's travelling to the prime tower location helps. Thanks!

closed 2024-12-07 by github-actions[bot]