Ultimaker/Cura #6663: Cannot re-preview UM S5 gcode / ufp

BagelOrb, 2019-11-15:
**Application version**
latest

**Platform**
Linux mint

**Printer**
S5

**Reproduction steps**
1. slice
2. save toolpaths
3. load model and quickly delete it (workaround for another bug which causes the gcode preview not to be flushed when switching to a different model)
4. load in .gcode or .ufp saved in step 2

**Actual results**
Preview shows massive overextrusion

**Expected results**
Preview the same as after step 1

**Additional information**
project, gcode, ufp
[S5_dual_preview.zip](https://github.com/Ultimaker/Cura/files/3852263/S5_dual_preview.zip)

BagelOrb, 2019-11-15:
Free screenshot:
[image]

Ghostkeeper, 2019-12-03:
I'm unable to reproduce this problem. I tried your reproduce steps (slice, save, load) and it displayed fine. Also the g-code files from your .zip archive seem to display normally:
[image]

Has this been fixed in the meanwhile? Could you try on the latest master?

BagelOrb, 2019-12-03:
Still reproduces on latest master for:
- UMS5_cube_2mm_dual.gcode
- UMS5_dual_brim.gcode
- UMS5_dual_brim.ufp

(PS: I can also reproduce it with the 3mf file, but I didn't update CuraEngine, so that test was inconclusive I guess.)

There's also some (other) update bug when loading gcode:
[image]

Ghostkeeper, 2019-12-05:
Seems I just tried with the wrong g-code file from your .zip. Indeed with the 2mm_dual.gcode it reproduces. Somehow it seems to go wrong on layer 18 there on the first extrusion move. This is the g-code involved:

```
;TYPE:SKIN
G1 F1260 X160.644 Y120.551 E16.03071
M204 S5000
M205 X50 Y50
G0 F9000 X159.559 Y119.466
M204 S500
M205 X5 Y5
G1 F1260 X159.458 Y119.365 E16.0311
M204 S5000
M205 X50 Y50
G0 F9000 X159.469 Y119.271
;MESH:NONMESH
G0 F600 X159.469 Y119.271 Z1.9
G0 F9000 X159.413 Y120.001
;TIME_ELAPSED:92.302087
;LAYER:17
M204 S500
M205 X5 Y5
;TYPE:WALL-OUTER
;MESH:cube_10cm.stl(1)
G1 F1200 X160 Y119.409 E16.03567  ; <-- This line seems to be huge.
G1 X159.902 Y119.175 E16.03706
G1 X160 Y119.175
G1 X160.588 Y119.175 E16.04029
```

Ghostkeeper, 2019-12-05:
Perhaps something goes wrong with the detected layer height?

BagelOrb, 2019-12-05:
That sounds like a likely candidate!

Perhaps it thinks the Z1.9 belongs to the previous layer somehow.

closed 2024-12-07 by github-actions[bot]