Ultimaker/Cura #6664: 4.4: Windows: Start menu entry in separate folder

thopiekar, 2019-11-16:
Just installed the installer I received from @rburema and I noticed the following.
Normally, when installing Cura on Windows, it will be placed in the Ultimaker Cura directory in the start menu.
However, using this installer it gets placed into a separate versioned directory like this:
[image]
I would expect the shortcut to be placed with all the other shortcuts. Additionally, it would just lokk very clean as before.

Ghostkeeper, 2019-12-03:
Yeah I noticed this as well using Windows 8.1.

Ellecross, 2019-12-04:
It started happening in Cura 4.3. Hehe
[image]
We'll take a look, maybe something not too hard to resolve.

Ellecross, 2019-12-04:
UM devs, see CURA-7029

thopiekar, 2019-12-16:
@Ellecross No problem! Thank you for bringing this onto your board! :+1: 
I'm doing good. Just busy more than ever. Hope you are doing well, too! :blush:

closed 2024-12-07 by github-actions[bot]