Ultimaker/Cura #6665: 4.4: UIX: Is it not a bit too much of Ultimaker?

thopiekar, 2019-11-16:
Don't get me wrong, but is that not getting too much now?
From my experience when talking with people about 3D printing and talking after a time about "Cura", it is clear for everyone that this piece of software is maintained, distributed and owned by Ultimaker.
But if I see what's happening in 4.4, I begin to get the feeling that you are feared to lose the identity of Cura. Are you sure that you need to worry that much?

Let's assume someone is using an Ultimaker printer and material, he will likely see exactly this on Windows:
[image]

You will see "Ultimaker" at four different places and especially in:

fieldOfView, 2019-11-16:
The logo change is mainly to get one consistent style across all Ultimaker products (compared to eg forum, cloud, contribute portal, etc)

cp2004, 2019-11-17:
I understand your concern that the 'Ultimaker' name is everywhere on your configuration, but that is because you have selected all ultimaker products. What do you suggest they put to tell the user what filament they are using, or which printer they are slicing for? If you called it PLA, then how would you distinguish between brands without mentioning the name?!

nallath, 2019-11-18:
On a completely personal note; I think that it was abundantly clear before we added the extra Ultimaker text that Cura was in fact made & paid for by Ultimaker. 

The main reason to change the logo in the top bar is indeed what fieldOfView mentioned; It's to make it consistent with all the other products. 

As for the other references to Ultimaker, that's mostly because you selected everything from Ultimaker that was out there. The material display is because it's just the brand and on the build plate is because it's actually there IRL.

thopiekar, 2019-11-18:
Well, while reading this, you are all right. It might be that I'm still too sensitive and still didn't get used to the name change (Cura -> Ultimaker Cura) actually. :roll_eyes:

closed 2019-11-18 by thopiekar