Ultimaker/Cura #6667: Cura 4.3.0 Windows 10 - Help -> Show Configuration Folder Not Working

Slimicus, 2019-11-16:
**Application version**
4.3.0

**Platform**
Windows 10

**Printer**
Ultimaker 2 Upgraded to 2+

**Reproduction steps**
1. Open Cura
2. Help -> Show Configuration Folder

**Actual results**
Nothing Happens

**Expected results**
Expecting a folder to open

**Additional information**

evtrados, 2019-11-18:
Hello @Slimicus,

I've tried reproducing your issue in several ways, but I always get a new windows explorer window with the configuration folder directory.
Could it perhaps be that the window does appear but it's out of focus (i.e., out of your monitor(s) borders)? Did you check Task Manager for a Windows explorer instance? If the window has indeed opened, there should be a Windows Explorer instance called simply "4.3" like in the screenshot below. If there is one, you can "right-click --> Switch to" the window.
[image]

Slimicus, 2019-11-18:
Here's a video showing the behavior, I checked before to make sure it wasn't out of focus, closed everything but Cura, restarted explorer, and rebooted too, but it never shows and under my processes nothing ever opens.

[b0RxZhTePm.zip](https://github.com/Ultimaker/Cura/files/3859545/b0RxZhTePm.zip)

evtrados, 2019-11-19:
Thanks for the video! That is interesting, indeed..
Could you perhaps send us your whole configuration folder (i.e., %appdata%\cura\) in a .zip archive, so that we try to reproduce the issue?

Slimicus, 2019-11-20:
Here you go,

[cura.zip](https://github.com/Ultimaker/Cura/files/3869397/cura.zip)

Ellecross, 2019-11-26:
What a very odd bug. :> I cannot reproduce it too, same as @evtrados 

What happens when you go to File > Open Files(s)? This part also open a windows dialogue. 
Have you tried this with [Cura 4.4](https://ultimaker.com/software/ultimaker-cura)?

evtrados, 2019-12-03:
Hi @Slimicus ,

Did you have any luck when using the latest v4.4 Cura release? Does it work for you there?
If not, could you maybe try using some similar functions in other programs in your Windows OS (i.e., options that will open a Windows Explorer window)? If all works as normal in the other programs then there is definitely something wrong with how Cura launches Explorer in that particular case.
Otherwise, I would suspect that your Windows Explorer might not be working properly when it comes to opening directories that are in "hidden" places, such as %appdata%. Actually, maybe you can try and change the View Options in Windows Explorer (if you haven't tried that already):
In a Windows Eplorer Window go to the View tab → Select Options on the right → Select the View tab in the pop up window → make sure the "Show hidden files, folders, and drives" is selected. It shouldn't make a difference, but you can try that anyway.

GDRPrint, 2020-11-06:
Same issue, Win 10 64 bit, Cura runs fine, no other issues that I know of. Clicking on "open config" produces nothing, no hidden windows etc. by checking Task mgr

hankster42, 2020-12-03:
I'm having the same issue but only on one desktop Win10 PC.  There are 3 versions of Cura present on this computer.  Cura 3.6.0 aka QidiPrint, Cura 4.2 aka Creality Splicer, and Cura 4.8.  All 3 versions exhibit the same result, when Show Configuration Folder is selected, nothing happens.  But my other Win10 PC which only has Cura 4.8 loaded works just fine.  Both PC's are running indentical copies of Win 10 Pro.  I've also compared settings on both PC's and they appear the same.

GDRPrint, 2020-12-03:
Only 1 copy running on my machine, thanks for your thoughts, hope someone takes it up.

konskarm, 2020-12-03:
I cannot reproduce the issue on Windows10. Can you check if the folder `%appdata%\cura\4.8` exists? If it does, could you provide the Cura logs `%appdata%\cura\4.8\cura.log` ?

hankster42, 2020-12-04:
Hope this is what you want
BTW  I found it at %appdata%/roaming/cura/4.8 if that matters
[cura.log](https://github.com/Ultimaker/Cura/files/5640453/cura.log)

konskarm, 2020-12-04:
If the path of the configuration folder could not be retrieved, an error `UnsupportedStorageTypeError` would appear in the logs, but that's not the case. Does it work in the previous Cura versions?
Although I do not expect anything different there, since the code that opens the configuration folder has not been touched since 2016.

Ghostkeeper, 2020-12-04:
> Does it work in the previous Cura versions?

Hankster42 mentions he's tested on forks of 3.6 and 4.2, and mainline 4.8.

Ghostkeeper, 2020-12-04:
For me it also still works fine in 4.8. None of the developers has been able to reproduce this issue so far.

konskarm, 2020-12-04:
> > Does it work in the previous Cura versions?
> 
> Hankster42 mentions he's tested on forks of 3.6 and 4.2, and mainline 4.8.

I missed, that, my apologies.

hankster42, 2020-12-04:
The fact that I experience no issue on my second PC makes me believe that the problem is with this one PC and the Win10 OS and NOT the application itself.  I just don't know where to look for and hoping for a clue.  The issue doesn't stop me from performing my slicing operation.

Ghostkeeper, 2020-12-07:
Perhaps there is a shell extension on that computer? Like, Google Drive or Dropbox, something that integrates with Windows Explorer?

dgrat, 2021-03-07:
No, I can reproduce the issue.

hankster42, 2021-03-07:
It's been a couple of months, and Win 10 has gotten at least one upgrade in that time.  I rechecked my Cura installs.  As before both Cura 4.8 and Creality Slicer which shows as Cura 4.2.1? still don't show Configuration Folder.  On my other Win 10 desktop (same version) I have Creality Slicer 4.2.1 and confirm it opens with no problem.

Ghostkeeper, 2021-03-15:
So far we haven't been able to reproduce this. Just tried again on Windows 10, but the file explorer opens file on the `%APPDATA%\Roaming\cura\4.8` folder.

phlnx, 2021-05-12:
I also have this issue. Running v4.9.1 (freshly updated) on Windows 10 19042.928. File -> Open Files shows the usual explorer/requester. Help -> Show Configuration Folder does absolutely nothing. Help -> About... opens the information popup.

All I wanted was to backup the configuration so that I could use it on a different machine without having to set it up again. I am grateful to those in this thread who provided the filesystem path to the configuration files.

laserbeak, 2021-11-26:
freezes for me when I click help -> show configuration folder. Cura 4.11.0

tfjield, 2022-06-14:
Not working for me with Cura 5.0.0 on Windows 10 64 bit.  I have 4 monitors, could that have something to do with it?  Cura can't seem to figure out which one I'm on when it brings up dialogs; they're all over the place.

GDRPrint, 2022-06-15:
I have multi monitore, no issues to report, I run Cura on the main screen.
Win 10 64 bit, fresh install which is a bit different than the older versions.
It ran fine on an old install from 2016 ish.

closed 2024-12-07 by github-actions[bot]