Ultimaker/Cura #6669: Cura Preview: allow you to change layer number to Z mm height

3dprinterhobby, 2019-11-16:
**Is your feature request related to a problem? Please describe.**
This would be an enhancement feature

**Describe the solution you'd like**
Currently in Preview you have the layer slider on the right hand side and the number equal the layer number you are viewing. It would be great if you could have a quick toggle allowing you to switch the number to represent the layer Z height.  Or even better than a toggle have it display both the layer # and the what that represents in layer height.

**Describe alternatives you've considered**
none.  other than burying a item into the bed with neg Z height

**Affected users and/or printers**
Where this really benefits is when a print fails, sometimes you can fix the part by just continuing at the same layer where it stopped.  My printer does not display the layer number it is on but rather Z height.  So having the ability to have the Z height visible will help to figure out where you need to start again.  The only other solution is to use a Caliper Measure tool.

**Additional context**
None

nallath, 2019-11-18:
There is some workaround for this; there is a post proccessing script that will make your printer show layer numbers during the print.

3dprinterhobby, 2019-11-18:
> There is some workaround for this; there is a post proccessing script that will make your printer show layer numbers during the print.

I should have mentioned I am running a CR10s printer, does the script you are referring to work with this printer?

nallath, 2019-11-19:
It should if it supports the M117 g-code (it should, since it's marlin based)

Ghostkeeper, 2019-12-04:
Duplicate of https://github.com/Ultimaker/Cura/issues/2536.

closed 2019-12-04 by Ghostkeeper