Ultimaker/Cura #6670: [4.3.0] Missing buttons on "Open project file" dialog

jjbott, 2019-11-17:
**Application version**
4.3.0

**Platform**
Windows 10

**Printer**
Creality-Ender-3

**Reproduction steps**
1) Open a 3mf file
2) A dialog box pops up asking how to open it

**Actual results**
[image]

**Expected results**
(What should happen after the above steps have been followed.)
[image]

**Additional information**
Seems to be caused by how Cura is handling (or isn't handling) Window's DPI settings. I have my scaling set to 225%. I got the "Expected results" screenshot by setting it to 100%
[image]

nallath, 2019-11-18:
So if that screen could be rescaled, would that fix your issue? 

It's almost impossible for us to build the interface in such a way that every screen is still usable if the text is set at 225%

jjbott, 2019-11-19:
It's not the text alone that's set to 225% - thats the DPI setting for *everything*. I think that difference is the core of the problem. 

It looks like Cura is scaling (some) UI elements depending on DPI settings, but it is not scaling the dialog itself. Notice above the 2 dialogs are the exact same size (ignoring the title bar). If it was respecting the DPI settings, one would be 225% wider and taller.

jjbott, 2019-11-19:
Another quick example to show how the UI elements are being correctly scaled based on DPI, but the dialog size is not:

100%:
[image]

225% - default size:
[image]

225% - Manually resized to roughly 225% width and height (although Github is probably not displaying it at full size):
[image]

nallath, 2019-11-19:
Well, we do seem to take screen scale factor into account when calculating the size of the window (which is something that can be seen by the fact that on 100%, it should work on all kinds of screens). 

I will have a look at this to see if I can tweak this behavior a bit.

nallath, 2019-11-19:
I've made some changes that should fix this issue, but I'm not on windows myself, so I can't test the actual solution.

jjbott, 2019-11-20:
Seems fine so far! Would you have to update the remaining theme.json files as well? If so, what a pain...

[image]

jjbott, 2019-11-20:
Side note, for fixed-sized dialogs like "Open project file", it'd be nice to add the Qt.MSWindowsFixedSizeDialogHint flag to the dialog. Without it, the dialog looks resizable (the mouse pointer changes to resizing arrows when hovering over the borders)

I did some experimenting, and looks like something like:
`flags: Qt.MSWindowsFixedSizeDialogHint | Qt.WindowTitleHint | Qt.WindowCloseButtonHint | Qt.WindowModal | Qt.Dialog`
would do the trick. Not sure if there is a way to avoid explicitly setting all of the default flags when you only want to add one...

Buuuut thats getting picky. :) I may have to do some experimenting and make my own MR someday...

Ghostkeeper, 2019-12-04:
> Seems fine so far! Would you have to update the remaining theme.json files as well? If so, what a pain...

No, theme files inherit from each other. The base is Cura Light theme, so the rest of the themes automatically take over the new theme entries and don't need to define this new size individually.

We'll consider this issue fixed then. Though yeah we could probably do something about that resize cursor on Windows...

closed 2019-12-04 by Ghostkeeper