Ultimaker/Cura #6671: Change at Z gcode changes are not shown in preview

photodude, 2019-11-17:
**Is your feature request related to a problem? Please describe.**
Change at Z gcode changes are not shown in preview

**Describe the solution you'd like**
It would be very helpful to see Change at Z gcode changes in preview

**Describe alternatives you've considered**
none

**Affected users and/or printers**
any user who makes gcode modifications with Change at Z would benefit in seeing those changes in the print preview

**Additional context**
This could also be considered a bug as it seems preview fails to consider post-processing of gcode in the preview. Cura 4.3
1) load model
2) add 3 speed changes
3) slice
4) preview feed rate
notice speed changes are not part of the preview
[image]

nallath, 2019-11-18:
At the moment that the preview is visible, the changes to the g-code are not yet made. The post-processing of the g-code happens right before you save it. As such, it's not really possible to show those changes without making pretty major changes to the way that we handle this.

photodude, 2019-11-19:
@nallath without making a major change, would it be possible to do a "post preview" update to the preview?

A process like the following

1. slice model
2. open preview
3. click button "update preview with post processing changes"
4. view updated preview

nallath, 2019-11-20:
Yes; You can save the g-code and load the saved g-code into Cura. At that point you are looking at the modified g-code.

photodude, 2019-11-20:
@nallath That's a good workaround to be able to see the preview with the modifications. But loading gcode to do the preview means I can't change something and quickly see how the printing would proceed.

The ideal case would allow preview without saving the gcode and then loading a gcode model 

What I was suggesting was a button/script which would trigger something to change the preview to include the post-processing gcode changes in the preview. I only suggested that as a possible solution as you said there would need to be some "pretty major changes" to get preview to show the gcode changes which are applied in the save process.

Ghostkeeper, 2019-12-04:
> But loading gcode to do the preview means I can't change something and quickly see how the printing would proceed.

Yeah but that's the real problem for implementation as well. Loading that g-code file in takes a lot of time to parse the g-code. But this is necessary because the post-processing script can make arbitrary changes to the g-code, so you'd need to re-parse everything.

Normally after slicing, CuraEngine sends its intermediary results to the front-end when they are still polygons, to display as layer view. It's not just sent as g-code, but also as polygons so that the front-end doesn't need to calculate the shapes again. That's why it takes only 2 seconds to display layer view after slicing, instead of 2 minutes.